openwrt/routing #79: nat46 compile broken with linux-3.18

mantas-p, 2015-02-19:
nat46 fails to compile after update to linux 3.18 (http://buildbot.openwrt.org:8010/broken_packages/ar71xx/nat46/compile.txt)

It seems that since linux-3.17 signature of ```alloc_netdev(sizeof_priv, name, setup)``` changed to  ```alloc_netdev(sizeof_priv, name, name_assign_type, setup)```

 Easy fix: http://rglinuxtech.com/?p=1184

sbyx, 2015-02-20:
Thanks for the report.
@ayourtch could you add a macro conditional so it works with both versions?

ayourtch, 2015-02-20:
I've committed the https://github.com/ayourtch/nat46/commit/a28e24b45c31bfa61447d0bdd591478cff1cad10
 Compiles OK on 3.12 and 3.19.

sbyx, 2015-02-21:
Thanks, I bumped the package. I guess this should be fixed then.

closed 2015-02-21 by sbyx