openwrt/routing #83: miniupnpd randomly crashes. no errors in log.

Manp82, 2015-03-16:
using hnyman build from https://forum.openwrt.org/viewtopic.php?id=28392 on a WNDR3700v2.

miniupnpd_1.9.20141209-1 seems to randomly crash with no errors left in the log, the process just disappear from the processes list. tested with default configuration. 

reverting back to miniupnpd_1.8.20140523-4 from https://downloads.openwrt.org/barrier_breaker/14.07/ar71xx/generic/packages/routing/ seems to be fixing the problem. miniupnpd has been running fine for the past 3 days, while 1.9.20141209-1 could never run for so long for me. 

would be possible to to bump miniupnpd to the latest version 1.9.20150307 to test it?

Manp82, 2015-04-01:
updated to 1.9.20150307 today. after 11 hours of uptime upnpd seems to be fine.
could never get that much on 1.9.20141209.

sbyx, 2015-04-07:
thanks for the update.

closed 2015-04-07 by sbyx

Manp82, 2015-04-08:
i can confirm, not a single issue since upgrading to 1.9.20150307-1.

thanks