openwrt/routing #96: for-15.05 release branch created

sbyx, 2015-05-04:
Hi @openwrt-routing/owners  ,

I just created the release-branch "for-15.05" for Chaos Calmer.
As with for-14.07 the guidelines for "Release Branches" from https://github.com/openwrt/packages/blob/master/CONTRIBUTING.md apply.

Thanks,

Steven

lgierth, 2015-05-04:
Thanks for the heads-up -- is there a rough schedule as to when the openwrt.git repo for 15.05 will be created, and the release candidate published?

sbyx, 2015-05-05:
The branch will be created after RC1 is out, there is at the moment some internal testing going on.

sbyx, 2015-06-14:
@openwrt/package-maintainers @openwrt/owners please get all pending fixes and security-updates into the for-15.05 branch in the following days. RC2 is officially out now and if we do not encounter any serious issue we might skip RC3 and go to final directly.

If there is any blocker for you or any issue that should be dealt with beforehand, please let us know.

sbyx, 2015-06-14:
@openwrt-routing/owners  please get all pending fixes and security-updates into the for-15.05 branch in the following days. RC2 is officially out now and if we do not encounter any serious issue we might skip RC3 and go to final directly.

If there is any blocker for you or any issue that should be dealt with beforehand, please let us know.

closed 2015-08-13 by bittorf