openwrt/routing #120: miniupnpd: multicast packets on wifi interfaces that were brought down causes the daemon to spam the logs with errors at an interval

braveheartleo, 2015-08-06:
It looks like miniupnpd (if `enable_upnp=yes`) would need a hotplug event that handles `ifdown` events for interfaces on which it was setup to listen on (log truncated):

```
Thu Aug  6 06:38:25 2015 daemon.err miniupnpd[5527]: sendto(udp_notify=18, [fcad:b10c:0:84::1]): Permission denied
Thu Aug  6 06:38:25 2015 daemon.err miniupnpd[5527]: sendto(udp_notify=18, [fcad:b10c:0:84::1]): Permission denied
Thu Aug  6 06:38:25 2015 daemon.err miniupnpd[5527]: sendto(udp_notify=19, 192.168.81.1): Invalid argument
Thu Aug  6 06:38:25 2015 daemon.err miniupnpd[5527]: sendto(udp_notify=19, 192.168.81.1): Invalid argument
Thu Aug  6 06:38:25 2015 daemon.err miniupnpd[5527]: sendto(udp_notify=20, [fcad:b10c:0:84::1]): Permission denied
Thu Aug  6 06:38:25 2015 daemon.err miniupnpd[5527]: sendto(udp_notify=20, [fcad:b10c:0:84::1]): Permission denied
Thu Aug  6 06:38:25 2015 daemon.err miniupnpd[5527]: try_sendto failed to send 3 packets
Thu Aug  6 06:38:25 2015 daemon.err miniupnpd[5527]: try_sendto(sock=12, len=469, dest=[ff02::c]:1900): sendto: Permission denied
Thu Aug  6 06:38:25 2015 daemon.err miniupnpd[5527]: try_sendto(sock=12, len=410, dest=[ff02::c]:1900): sendto: Permission denied
Thu Aug  6 06:38:25 2015 daemon.err miniupnpd[5527]: try_sendto failed to send 2 packets
Thu Aug  6 06:38:25 2015 daemon.err miniupnpd[5527]: try_sendto(sock=12, len=449, dest=[ff02::c]:1900): sendto: Permission denied
Thu Aug  6 06:38:25 2015 daemon.err miniupnpd[5527]: try_sendto(sock=12, len=410, dest=[ff02::c]:1900): sendto: Permission denied
Thu Aug  6 06:38:25 2015 daemon.err miniupnpd[5527]: try_sendto failed to send 2 packets
Thu Aug  6 06:38:25 2015 daemon.err miniupnpd[5527]: try_sendto(sock=12, len=410, dest=[ff02::c]:1900): sendto: Permission denied
Thu Aug  6 06:38:25 2015 daemon.err miniupnpd[5527]: try_sendto failed to send 1 packets
Thu Aug  6 06:38:25 2015 daemon.err miniupnpd[5527]: try_sendto(sock=12, len=465, dest=[ff02::c]:1900): sendto: Permission denied
Thu Aug  6 06:38:25 2015 daemon.err miniupnpd[5527]: try_sendto(sock=12, len=465, dest=[ff02::c]:1900): sendto: Permission denied
Thu Aug  6 06:38:25 2015 daemon.err miniupnpd[5527]: try_sendto(sock=12, len=463, dest=[ff02::c]:1900): sendto: Permission denied
Thu Aug  6 06:38:25 2015 daemon.err miniupnpd[5527]: try_sendto(sock=12, len=481, dest=[ff02::c]:1900): sendto: Permission denied
Thu Aug  6 06:38:42 2015 daemon.err miniupnpd[5527]: sendto(udp_shutdown=8): No such device
Thu Aug  6 06:38:42 2015 daemon.err miniupnpd[5527]: sendto(udp_shutdown=8): No such device
Thu Aug  6 06:38:42 2015 daemon.err miniupnpd[5527]: sendto(udp_shutdown=8): No such device
Thu Aug  6 06:38:42 2015 daemon.err miniupnpd[5527]: sendto(udp_shutdown=8): No such device
Thu Aug  6 06:38:42 2015 daemon.err miniupnpd[5527]: sendto(udp_shutdown=8): No such device
Thu Aug  6 06:38:42 2015 daemon.err miniupnpd[5527]: sendto(udp_shutdown=8): No such device
Thu Aug  6 06:38:42 2015 daemon.err miniupnpd[5527]: sendto(udp_shutdown=8): No such device
Thu Aug  6 06:38:42 2015 daemon.err miniupnpd[5527]: sendto(udp_shutdown=8): No such device
Thu Aug  6 06:38:42 2015 daemon.err miniupnpd[5527]: sendto(udp_shutdown=8): No such device
Thu Aug  6 06:38:42 2015 daemon.err miniupnpd[5527]: sendto(udp_shutdown=8): No such device
Thu Aug  6 06:38:42 2015 daemon.err miniupnpd[5527]: sendto(udp_shutdown=8): No such device
Thu Aug  6 06:38:42 2015 daemon.err miniupnpd[5527]: sendto(udp_shutdown=12): Permission denied
Thu Aug  6 06:38:42 2015 daemon.err miniupnpd[5527]: sendto(udp_shutdown=12): Permission denied
Thu Aug  6 06:38:42 2015 daemon.err miniupnpd[5527]: sendto(udp_shutdown=12): Permission denied
Thu Aug  6 06:38:42 2015 daemon.err miniupnpd[5527]: sendto(udp_shutdown=12): Permission denied
Thu Aug  6 06:38:42 2015 daemon.err miniupnpd[5527]: sendto(udp_shutdown=12): Permission denied
Thu Aug  6 06:38:42 2015 daemon.err miniupnpd[5527]: sendto(udp_shutdown=12): Permission denied
Thu Aug  6 06:38:42 2015 daemon.err miniupnpd[5527]: sendto(udp_shutdown=12): Permission denied
Thu Aug  6 06:38:42 2015 daemon.err miniupnpd[5527]: sendto(udp_shutdown=12): Permission denied
Thu Aug  6 06:38:42 2015 daemon.err miniupnpd[5527]: sendto(udp_shutdown=12): Permission denied
Thu Aug  6 06:38:42 2015 daemon.err miniupnpd[5527]: sendto(udp_shutdown=12): Permission denied
Thu Aug  6 06:38:42 2015 daemon.err miniupnpd[5527]: sendto(udp_shutdown=12): Permission denied
```

This is what happens when I turned off my wireless radios (2.4 Ghz with 2 VAPs, 5 Ghz with 2 VAPs) via a button press or `wifi down` at the cli. These four interfaces are all routed, and miniupnpd continuously spams the logs with these errors depending on the `notify_interval` value.

This is an issue because miniupnpd tries to send those multicast packets on non-existent interfaces, but fails to do so. This is not an issue if miniupnpd is started with `enable_upnp=no`.

braveheartleo, 2015-08-10:
It turns out that the fix is simply modifying `/etc/hotplug.d/iface/50-miniupnpd` and commenting out the line:
```
#[ ! "$ACTION" = "ifup" ] && service_check /usr/sbin/miniupnpd && exit 0
```

This way, miniupnpd is restarted as soon as interfaces specified in `internal_iface` are either brought up or down. This takes care of the problem that when an interface is brought down, miniupnpd would spam the logs continuously and complain about multicast packets it was unable to send on the interface, as was the issue of this bug report.

ecsv, 2019-05-27:
This package was removed in 20987707105119f50487711682bb269762ef43c9. So this ticket should most likely be closed