openwrt/routing #131: babeld: unable to specify bridge as interface

tcatm, 2015-08-14:
It is not possible to use a bridge as an interface in babeld (when configured using UCI).

A simple

    config interface 'wan'

will resolve `wan` to the first interface of that bridge (in my case `eth0`) as opposed to `br-wan`. This, obviously, breaks in funny ways.

The sample UCI config states to use the full interface name instead to avoid this mess, like this:

    config interface 'br-wan'

Well, that doesn't work either. A dash is not allowed by UCI in section names.

Can we restructure the config to make `ifname` an option of the configuration section?

kerneis, 2015-08-17:
Out-of-curiosity, why did you close this? It looks like a serious issue
that should be fixed (although I won't have time to look into this in
the next few weeks, I'd be happy to review a pull request).

tcatm, 2015-08-17:
Oh, for some reason I was using a very old version of the babeld package. Newer versions do have this exact bug fixed in exactly the way I suggested. Sorry!

kerneis, 2015-08-17:
> Oh, for some reason I was using a very old version of the babeld package. Newer versions do have this exact bug fixed in exactly the way I suggested. Sorry!

It would still be better if we managed to handle bridges aliases correctly too. Does that work with the newer version? If it is not doable, we should at least add a comment warning against aliases for bridges in the sample UCI file (there is nothing about it currently).

tcatm, 2015-08-17:
The current init-script does not handle aliases at all and accepts raw interface names only. I'm not sure whether that's a good solution but it works fine with the way babeld handles interfaces.

kerneis, 2015-08-17:
I'm not sure I follow you: we call network_get_device in babeld.init.
https://github.com/openwrt-routing/packages/blob/master/babeld/files/babeld.init#L178

And we claim to support aliases:
https://github.com/openwrt-routing/packages/blob/master/babeld/files/babeld.config#L31

Maybe zorun@ can provide more inisght into this?

tcatm, 2015-08-18:
I've just checked with config/network:

    config interface wan
       type bridge
       ifname eth0.1

and config/babeld:

    config interface
        config ifname wan

Does not work and writes "interface wan" to the generated config file.

zorun, 2015-08-22:
This looks related to https://github.com/openwrt-routing/packages/pull/52#issuecomment-55004440

AFAICT, logical interface name resolution never worked very reliably.  We should either find a new method, or drop support for logical interface name altogether.