openwrt/routing #135: Nodogsplash config problem

netomx, 2015-09-01:
Hi!
When you try to enable PasswordAuthentication, the config file will just output:
<pre>
PasswordAuthentication
</pre>
Instead of:
<pre>
PasswordAuthentication yes
</pre>
Modifying line 72 and 78 will make it work again.
Also, it will repeat twice some configs:
<pre>
# auto-generated config file from /etc/config/nodogsplash
GatewayInterface br-lan
gatewayname Invitado
maxclients 10
clientforcetimeout 10
passwordattempts 3
passwordauthentication yes
passwordattempts 3
password snack
AuthenticateImmediately yes
PasswordAuthentication yes
Password snack
</pre>

lynxis, 2016-06-02:
@mwarning 
I would say we have to remove some lines:

https://github.com/openwrt-routing/packages/blob/master/nodogsplash/files/nodogsplash.init#L210
https://github.com/openwrt-routing/packages/blob/master/nodogsplash/files/nodogsplash.init#L211
https://github.com/openwrt-routing/packages/blob/master/nodogsplash/files/nodogsplash.init#L213
https://github.com/openwrt-routing/packages/blob/master/nodogsplash/files/nodogsplash.init#L214

netomx, 2016-06-02:
Why do you say that? They are valid options

mwarning, 2016-06-02:
@netomx I think v2  does not support those right now. I probably needs to be added if there is need (an motivation to add it)

netomx, 2016-06-02:
I didnt know there was a v2 nodogsplash. Since when?

mwarning, 2016-06-02:
@netomx it was merged into master today. Release will probably be "soon", but it does not have all features of 1.0.0.

lynxis, 2016-06-05:
@netomx @mwarning those options are parsed in a their own function in the script:
https://github.com/openwrt-routing/packages/blob/master/nodogsplash/files/nodogsplash.init#L63

mwarning, 2016-06-05:
@lynxis ah, now I understand what you want :-) I will remove duplicate lines.

mwarning, 2016-06-05:
@lynxis it is probably better to just remove setup_user_authentication(), otherwise, PasswordAuthentication/PasswordAuthentication cannot be set individually.

commit f3f344e9 (2016-06-05): nodogsplash: remove duplicate options, fixes #135
closed 2016-06-05 by mwarning