openwrt/routing #141: [olsrd] jsoninfo does not have a valid config

stargieg, 2015-11-10:
the generated olsrd6 olsrd_jsoninfo.so.0.0 config section is invalid
Workaround: a second run off /etc/init.d/olsrd6 restart

uci show olsrd6.@LoadPlugin[4]

```
olsrd6.cfg124e54=LoadPlugin
olsrd6.cfg124e54.accept='::'
olsrd6.cfg124e54.ipv6only='yes'
olsrd6.cfg124e54.library='olsrd_jsoninfo.so.0.0'
olsrd6.cfg124e54.ignore='0'
```

cat /var/etc/olsrd6.conf

```
LoadPlugin "olsrd_jsoninfo.so.0.0"
{
	PlParam "name" "g15-nsm2"
	PlParam "lat" "52.0"
	PlParam "lon" "13.0"
	PlParam "accept" "::"
	PlParam "ipv6only" "yes"
	PlParam "services-file" "/var/etc/services.olsr.ipv4"
	PlParam "suffix" ".olsr"
	PlParam "latlon-file" "/var/run/latlon.js.ipv4"
	PlParam "hosts-file" "/tmp/hosts/olsr.ipv4"
}
```

I will debug the init script next time

stargieg, 2015-11-10:
Interesting
olsr v4 LoadPlugin[3] has the same hash as olsr v6 LoadPlugin[4]

uci show olsrd.@LoadPlugin[3]
```
olsrd.cfg124e54=LoadPlugin
olsrd.cfg124e54.services_file='/var/etc/services.olsr.ipv4'
olsrd.cfg124e54.library='olsrd_nameservice.so.0.3'
olsrd.cfg124e54.suffix='.olsr'
olsrd.cfg124e54.latlon_file='/var/run/latlon.js.ipv4'
olsrd.cfg124e54.hosts_file='/tmp/hosts/olsr.ipv4'
```
uci show olsrd6.@LoadPlugin[4]
```
olsrd6.cfg124e54=LoadPlugin
olsrd6.cfg124e54.accept='::'
olsrd6.cfg124e54.ipv6only='yes'
olsrd6.cfg124e54.library='olsrd_jsoninfo.so.0.0'
olsrd6.cfg124e54.ignore='0'
```

bittorf, 2015-11-11:
when reading http://www.olsr.org/?q=jsoninfo_plugin i dont see e.g. lat/lon/hosts-file.
do you mix it up with nameserver-plugin?

stargieg, 2015-11-11:
no. In my First Post is the olsr6 jsoninfo config section.
The lat/lon/hosts-file come from the olsr4 nameservice section.
There is an issue in the name space of the init script.

bittorf, 2015-11-11:
@stargieg ok...understand. can you post your "uci show olsrd" so we can better understand it.

stargieg, 2015-11-11:
Hi Bastian
I think you mean uci show olsrd6
You can trigger this issue by change network settings and add hna6 to olsr6 config.
I have mask my private ip6 net with "ffff"

[config-olsrd.txt](https://github.com/openwrt-routing/packages/files/32150/config-olsrd.txt)
[tmp-olsrd.conf.txt](https://github.com/openwrt-routing/packages/files/32148/tmp-olsrd.conf.txt)
[config-olsrd6.txt](https://github.com/openwrt-routing/packages/files/32149/config-olsrd6.txt)
[tmp-olsrd6.conf.txt](https://github.com/openwrt-routing/packages/files/32151/tmp-olsrd6.conf.txt)

bittorf, 2015-11-11:
@stargieg: can you give me SSH-access to the box? - please poke me at bastian@jabber.weimarnetz.de

bittorf, 2015-11-23:
if i understand 96fb4b6aebc1f26b66d1a03bebca0c05bc2380dd correctly, this is not olsrd-bug, but luci2 - so i will close this as "not a bug"

closed 2015-11-23 by bittorf