openwrt/routing #148: olsrd.hotplug.sh doesn't work on interface aliases

dangowrt, 2016-02-12:
On a router, several IPv4 subnets are configured on the same LAN interface. One of them is used for olsrd, I tried both ways of setting up the interface:
```
config interface 'olsr'
    option ifname '@lan'
    option proto 'static'
    option ipaddr '10.x.x.x'
    option netmask '255.255.255.255'
```
as well as the legacy way
```
config alias 'olsr'
	option proto 'static'
	option interface 'lan'
	option ipaddr '10.x.x.x.x'
	option netmask '255.255.255.255'
```
In both cases, the olsrd.hotplug.sh script doesn't correctly detect that the Linux interface belongs to a specific OpenWrt UCI interface. This results in removing and re-adding the interface every 2.5s, ie. flapping routes and lots of log-spam.

I suggest that the ```olsrd.hotplug.sh``` shell script should be fixed to correctly resolve interface aliases.

dangowrt, 2016-02-12:
@zioproto @sbyx

bittorf, 2016-02-12:
can you please post the logread-output?
i will dig into it  - thank you!

dangowrt, 2016-02-12:
```
Fri Feb 12 14:17:46 2016 daemon.info olsrd[18220]: olsr.org - 0.9.0.3-git_0000000-hash_b3d5b17376aa54962791415a01f0e399 successfully started
Fri Feb 12 14:17:48 2016 daemon.info olsrd[18220]: Removing interface lan
Fri Feb 12 14:17:48 2016 daemon.info olsrd[18220]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/lan/send_redirects
Fri Feb 12 14:17:48 2016 daemon.info olsrd[18220]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/lan/rp_filter
Fri Feb 12 14:17:48 2016 daemon.info olsrd[18220]: Adding interface lan
Fri Feb 12 14:17:50 2016 daemon.info olsrd[18220]: Removing interface lan
Fri Feb 12 14:17:50 2016 daemon.info olsrd[18220]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/lan/send_redirects
Fri Feb 12 14:17:50 2016 daemon.info olsrd[18220]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/lan/rp_filter
Fri Feb 12 14:17:50 2016 daemon.info olsrd[18220]: Adding interface lan
Fri Feb 12 14:17:52 2016 daemon.info olsrd[18220]: Removing interface lan
Fri Feb 12 14:17:52 2016 daemon.info olsrd[18220]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/lan/send_redirects
Fri Feb 12 14:17:52 2016 daemon.info olsrd[18220]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/lan/rp_filter
Fri Feb 12 14:17:52 2016 daemon.info olsrd[18220]: Adding interface lan
```
and so on... reliably repeats every 2.5s.

/etc/config/network looks like this:
```
config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config device 'modemvlan'
	option ifname 'eth0'
	option name 'modem'
	option type '8021q'
	option vid '7'

config device 'lanvlan'
	option ifname 'eth0'
	option name 'lan'
	option type '8021q'
	option vid '1'

config interface 'lan'
	option ifname 'lan'
	option proto 'static'
	option defaultroute '0'
	option peerdns '0'
	option AdvRouterAddr '1'
	option ipaddr '192.168.0.138'
	option netmask '255.255.255.0'

config interface 'wan'
	option proto 'pppoe'
	option ifname 'modem'
	option username 'XXX'
	option password 'YYY'

config interface 'modem'
	option proto 'static'
	option ifname 'modem'
	option netmask '255.255.255.0'
	option defaultroute '0'
	option peerdns '0'
	option ipaddr '10.0.0.1'

config interface 'olsr'
	option proto 'static'
	option ifname '@lan'
	option ipaddr '10.81.12.62'
	option netmask '255.255.255.255'
```

The  problem disappears if I simply make the 'olsr' interface be the main one and 'lan' an alias referring to '@olsr'. It's kinda obvious to me that the ```olsrd.hotplug.sh``` doesn't resolve interface aliases...

bittorf, 2016-02-12:
strange: i applied your config here and it is working fine:
olsrd.hotplug.sh doesnt look for an alias and there is no need for it. it just looks if the interface given via hotplug is already written to config. die config-writer in /lib/functions/olsrd.sh uses e.g. network_get_device IFNAME "$interface" and works with aliases...can you give me SSH access to the device?

dangowrt, 2016-02-12:
I just realize that having the OLSR address as the _primary_ address of the interface is not an option, because odhcpd doesn't support assigning addresses belonging to the subnet of any secondary aliases (and that kinda makes sense, because there can anyway be only one DHCP server on any layer-2 interface)...

dangowrt, 2016-02-12:
Ok, that's weird.
Please note that the 'lan' interface is a VLAN, and we got several IP aliases sitting on that single VLAN, thus the combination of VLANs and IP-aliases might be the cause. To me it seems like it correctly adds the interface to OLSR, but then removes it again because it thinks that the Linux ifname in the generated olsrd.conf doesn't belong to any interface configured to be used by OLSR in UCI... Once removed, it will add it again, because it's obviously missing...
I'm running CC 15.05.1 on x86-64.

dangowrt, 2016-02-12:
I sent an email to you, please reply with an SSH public key so I can give you remote access

dangowrt, 2016-02-13:
Ok, this seems to be a problem with olsrd itself rather than with the hotplug scripts.
I reported it upstream
https://lists.olsr.org/pipermail/olsr-dev/2016-February/008131.html

closed 2016-02-13 by dangowrt