openwrt/routing #170: dependencies for map-t (kmod-ipv6)

jallakim, 2016-04-10:
Hi,

In december there was two commits that related to `kmod-ipv6` changing to `@IPV6`;
* https://dev.openwrt.org/ticket/21358
* Commit bcaa87776f2eb2eb54675b2dca152d23625089ad (changing from `+kmod-ipv6` to `+@IPV6`).
* Commit 5cd413d16a802f0257cf92d0b5772d9b60863a3a (basically reverting the previous commit)

I tried installing `map-t` from trunk, but got dependency errors. Created a ticket, https://dev.openwrt.org/ticket/22201, but then realized the separation of the package `map` and `map-t` (where the latter is a metapackage of `nat46`). I guess creating an issue here is more suited.

As far as I can see, the change to `@IPV6` is correct, but the commit had `+` in front, which is the wrong syntax, as kan be seen in the `Makefile` for `map`; http://git.openwrt.org/?p=openwrt.git;a=blob;f=package/network/ipv6/map/Makefile;hb=HEAD

Maybe this could get changed, so that these packages work in trunk? (-:

dedeckeh, 2016-06-12:
Launched a PR (https://github.com/openwrt-routing/packages/pull/195) to fix the issue

dedeckeh, 2016-06-12:
Issue can be closed as commit https://github.com/openwrt-routing/packages/commit/85e771d2f1ca1981983238073ffc602981acbb40 fixes the issue

closed 2016-06-13 by jallakim