openwrt/routing #171: You can't add a wired link to a batadv mesh

diegoroversi, 2016-04-11:
It looks like there is no way to add a wired link to a mesh with batadv protocol. Or if exist, is not documented.

Documentation in https://wiki.openwrt.org/doc/howto/mesh.batman, show how to add a wifi-iface to the batadv mesh. But no example is given for a normal interface.

FreifunkUFO, 2016-06-05:
maybe the freifunk-gluon-firmware (with "mesh-on-wan / mesh-on-lan" feature) makes it nicer?
on freifunk-leipzig (via meshkit/imagebuilder) we do that since years with a crazy cronjob :-o
batctl -m bat12 if add eth0.1
batctl -m bat12 if add eth0.2

ecsv, 2018-06-13:
I would have said that it is rather easy because it should behave like any other network.interface. So adding an ifname will add it:

    config interface 'batnet'
            option mtu '1532'
            option proto 'batadv'
            option mesh 'bat0'
            option ifname eth1

Netifd is then taking care of adding eth1 (by calling proto_batadv_setup) to bat0 whenever eth1 finds an carrier and the device is also is up (in context of netifd). Most likely more adjustments for netifd are necessary to let it handle more than one interface in ifname (like it does for bridges).

ecsv, 2018-06-19:
It seems like it is also not possible to use ifname + at the same time the network information in /etc/config/wireless. Only the ifname will be used and wifi device is completely ignored.

It looks like the functionality was implemented by @mwarning,  @altergui and @lindnermarek in https://git.open-mesh.org/openwrt-feed-devel.git/commit/6bbb08cdf2a3126351646e77d1d1c6a1e77a9603

Maybe they can tell us how this is supposed to work.

ecsv, 2019-02-23:
No answer until now. The main confusion here is that the configuration is the wrong way around. So logically you should have a batadv_hardif proto for the slave devices and batadv (or batadv_meshif) proto for the configuration of the mesh interface. There is also a batadv_vlan proto for vlan configuration - but this part seems to be already implemented in an acceptable (logical) way.

But as it currently looks like, the batadv_hardif proto is actually called batadv. And there is no actual proto for the meshif. The latter has to be configured using /etc/config/batman-adv and requires an ugly hack via hotplug events. This unfortunately breaks various things - like network reloads for example.

That said, the current way to add multiple interfaces is using multiple interfaces with batadv proto:

    # configure ip on top of eth0
    config interface 'eth0_p2p_debug'
    	option ifname 'eth0'    
    	option proto 'static'
    	option ipaddr '192.168.2.1'
    	option netmask '255.255.255.0'
    	option ip6assign '60'
    
    # add all wireless interfaces with network bat0_hardif_wlan as hardif to bat0
    config interface 'bat0_hardif_wlan'
    	#option routing_algo 'BATMAN_IV'
    	option mtu '1536'
    	option proto 'batadv'
    	option mesh 'bat0'
    	#option ifname is filled out by the wireless config interface
    
    # add ifname from eth0_p2p_debug as hardif to bat0 
    config interface 'bat0_hardif_eth0'
    	#option routing_algo 'BATMAN_IV'
    	option proto 'batadv'
    	option mesh 'bat0'
    	option ifname '@eth0_p2p_debug'
    	option mtu '1536'
    
    # add eth1 as hardif to bat0
    config interface 'bat0_hardif_eth1'
    	#option routing_algo 'BATMAN_IV'
    	option proto 'batadv'
    	option mesh 'bat0'
    	option ifname 'eth1'
    	option mtu '1536'
    
    # add eth2 as hardif to bat0
    config interface 'bat0_hardif_eth2'
    	#option routing_algo 'BATMAN_IV'
    	option proto 'batadv'
    	option mesh 'bat0'
    	option ifname 'eth2'
    	option mtu '1536'
    
    # configure IP on bat0
    config interface 'batlan'
    	option ifname 'bat0'    
    	option proto 'static'
    	option ipaddr '192.168.1.1'
    	option netmask '255.255.255.0'
    	option ip6assign '60'
    
    # create vlan 23 on top of bat0 and enable ap_isolation for it
    config interface 'bat0_vlan_user'
    	option proto 'batadv_vlan'
    	option ifname 'bat0.23'
    	option ap_isolation '1'
    
    # configure ip on bat0.23
    config interface 'batuser_vlan'
    	option ifname '@bat0_vlan_user'    
    	option proto 'static'
    	option ipaddr '192.168.23.1'
    	option netmask '255.255.255.0'
    	option ip6assign '60'

This will add eth0, eth1, eth2 and wifi interface with network "bat0_hardif_wlan" to bat0. Make sure that routing_algo is always the same or the configured algo is random.

There is also a vlan (vid 23) interface on top of bat0 that has mesh-wide ap_isolation enabled. The interfaces eth0, bat0 and bat0.23 will receive an IP address

ecsv, 2019-05-24:
This was not possible with a single batadv proto interface. Multiple were required as shown in https://github.com/openwrt-routing/packages/issues/171#issuecomment-466624604

To reduce the confusion about this part, batadv was splitted in batadv (for the actual virtual interface) and batadv_hardif (the hardif which is added as slave to batadv). This can be found in 54af5a209e0a0a75b5eb712c0ca8056e66de02c0

The issue should be closed.

closed 2019-05-24 by simonwunderlich