openwrt/routing #172: miniupnpd defaults to UPnP 2.0

satlan, 2016-04-13:
Sometimes in 2015, miniupnpd moved to default to UPnP 2.0 which isn't correctly supported by Xbox One (the console fails to open any port forward, which prevents multiple xbox from working behind nat) 

See http://miniupnp.tuxfamily.org/forum/viewtopic.php?p=4573#4573
This was reverted in https://github.com/miniupnp/miniupnp/commit/58b130116ba7f1a3669cd822811822b4967c5ac7

Could miniupnpd in openwrt updated to reflect that change or moved to a newer miniupnpd version?

Or maybe openwrt could offer a separate package of UPnP 2 (since miniupnpd doesn't seem to be configurable in that respect).

Thanks.

AstarothPoD, 2017-06-28:
Any work on this? miniupnpd-package isn't working for consoles or Windows 10 since Upnp 2.

diizzyy, 2017-07-02:
v1 is deprecated and bad practice, if you want a legacy package feel free to submit a pull request or just compile it yourself. Better is to report to your vendor upstream instead.

AstarothPoD, 2017-07-03:
v1 is the only generally vendor supported version, which the original miniupnp project also realised a year and a half ago and reverted to 1.1 as linked above.
I suggest reverting or making it an option in UCI, because at the moment everyone that needs upnp support for gaming in any form have to recompile themselves, rendering this a pretty useless package.

diizzyy, 2017-07-03:
Because you only use upnp for gaming, did it every occur to you that it can other use cases?
A v1 and v2 package would be the proper solution in any.

ecsv, 2019-05-27:
This package was removed in 20987707105119f50487711682bb269762ef43c9. So this ticket should most likely be closed