openwrt/routing #184: Merging the openwrt-routing/packages with openwrt/packages?

jow-, 2016-05-04:
Hi @openwrt-routing/owners

for a long time both the openwrt-routing and openwrt packages organizations are working in parallel and in many cases are maintained by the same persons.

For the sake of simplicity I'd like to suggest to merge both organizations and move the packages from here into openwrt/packages.

This issue is just for starting a discussion and maybe reaching a consensus, so please voice your opinion here.

ffainelli, 2016-05-04:
Seems reasonable to me to merge this within openwrt/packages. AFAIR this project started earlier than when we decided to move the packages maintenance to Github, so no objection from me.

zioproto, 2016-05-04:
Hello @jow- 
sounds good to me to merge organizations. Less overhead is always better.
openwrt-routing was kind of a small group, where everyone knows each other because of the many OpenWrt and Battlemesh events.
If the group of committers gets wider in the new organization we should find a way to resolve the situation where there is a patch that needs further improvement before it gets accepted.

I have a proposal, can we set up the repositories in a way that is never possible to push directly but it is always necessary a pull request ? In this way we know that each change had at least 1 other person reviewing it.

lynxis, 2016-06-02:
@zioproto you can protect branches and enable require status checks. it's not the same, but nearby.

lgierth, 2016-06-02:
No objections from my side

mwarning, 2016-06-02:
sure, go ahead

bittorf, 2016-06-10:
i'am against it. it's easier for me to follow in this special field and i dont want to read all the messages from packages.

axn, 2016-06-10:
I see the point and for me both strategies are fine.

Could it mean less flexibility when interested in bleeding-edge routing
packages from trunk but stable openwrt packages? Now one could easily
choose different branches for each repo. But that's probably not so
relevant.

/axel

elektra42, 2016-06-10:
Hi –

> I see the point and for me both strategies are fine.
> 
> Could it mean less flexibility when interested in bleeding-edge routing
> packages from trunk but stable openwrt packages? Now one could easily
> choose different branches for each repo. But that's probably not so
> relevant.

+1

Cheers,
Elektra

-- 
Viral meme of radical freedom 
 
The fact that you talk in your head doesn't mean that you think. 
 
The best way to lose control over yourself is trying to control yourself. 
 
Most people experience themselves as a voice in their head, telling them 
 who they are, what they think and what they have to do. 
 
http://en.wikipedia.org/wiki/Meme

aparcar, 2017-09-04:
any progress on that?

diizzyy, 2017-09-04:
So far (using members listed by github): 

+1 jow, zioproto, mwarning, axn, elektra42
-1 bittrof

Give it a 2w timeout for the above to reply, no answer is counted as no opinion/blank. Seems reasonable and fair?

Missing: 
@p4u @lindnermarek @lgierth @altergui @acinonyx 
Any objections with merging?

aparcar, 2017-09-04:
@dangowrt @lynxis opinions?

lindnermarek, 2017-09-08:
Hi @jow- et al,

I am in the boat with @zioproto - as long as maintainers can require patch peer-reviews for specific packages I am fine with merging. We already had the case where a well-intended batman-adv patch pushed by a non-maintainer broke our package while we maintainers were unaware of said change until people complained about the broken package.

The protected branches mentioned by @lynxis does not sound like a solution to this problem. All packages live in subdirectories and not branches. Anybody else with a suggestion ?

@diizzyy: Not sure why @zioproto is counted as 'yes' since the problem he raised remains unsolved.

Cheers!

diizzyy, 2017-09-08:
@lindermarek: I think it's going to be hard to implement this as some packages are very use-case specific but I do agree there should be a pull request with some timeout for reviewing however for major changes. I don't see a need for minor changes however.

lindnermarek, 2017-09-08:
@diizzyy: That differentiation is very fuzzy. What constitutes a major changes / a minor change ?

In our case, the change was considered minor and broke the package nonetheless ..

diizzyy, 2017-09-08:
@lindnermarek 
Major: Major version jumps, additional patches that are intrusive etc?
Minor: Minor version jumps, configuration changes (that doesn't affect availability) functionality such as adding threading support. I would consider lets say removal of unicode for instance a major change. 

That said, we should always require a run-time test. I don't think there's a simple fix for this, reasonable would be that maintainer(s) should be able to commit directly while "external" pull requests should be reviewed within a reasonable time otherwise a commit should be allowed to step in and commit if requirements are fulfilled and it wont affect other packages

BKPepe, 2019-06-27:
Recently the package smcroute was moved from this repo to openwrt/packages.
See @lynxis's https://github.com/openwrt/packages/pull/8855#issuecomment-488138277

The advantages what I can see are that openwrt/packages has CircleCI, so the packages are compile tested against ath79 target and it has DCO check (Signed-off-by commits). It's not a big deal to have it here as well. Let's look, for example, at https://github.com/openwrt-routing/packages/pull/215 and we can think why the pull request wasn't tightened to the end.

However, I think in openwrt/packages there's a lot of activity.  This situation can be improved here if there will be more people with commit access or more people, who will maintain this repo.

mwarning, 2019-06-27:
Still, It would be nice to have a repository that has relaxed rules (e.g. for less polished packages).

diizzyy, 2019-06-27:
I presume the goal is to have as little breakage as possible so another other option would be to detach the routing repo if people don't feel they want to raise the bar slightly.

mwarning, 2019-06-27:
@diizzyy as the vote is already in favor of merging. How about doing that, and create another repository in the long run. But this decision is not part of this ticket.

aparcar, 2019-07-05:
I created a pull requests, basically moving all this to openwrt/packages/net, see here https://github.com/openwrt/packages/pull/9399

We could start using a [codeowners](https://help.github.com/en/articles/about-code-owners) file to define responsibilities per package.

BKPepe, 2019-07-05:
> We could start using a [codeowners](https://help.github.com/en/articles/about-code-owners) file to define responsibilities per package.

https://github.com/openwrt/packages/issues/7125

It will work just for people, which has commit access.

aparcar, 2019-07-05:
I don't see a problem giving the maintainers of routing access to packages, or would that be a problem?

adrianschmutzler, 2019-07-05:
@aparcar But it won't be available to non-maintainers, so those who were watching the whole openwrt-routing would now have to watch the whole openwrt/packages?

mwarning, 2019-07-05:
@adrianschmutzler those would likely only watch their own packages.

adrianschmutzler, 2019-11-04:
Hi, it looks like this is stuck again.
Based on the discussion in packages repository, the majority (out of very few people) favored moving packages without history. Despite, there was a preference (especially from the people at openwrt/packages) to not just merge packages, but put some work into them to meet the merge criteria for openwrt/packages: 
https://github.com/openwrt/packages/pull/9399#issuecomment-508727872

Unfortunately, almost nothing has happened since then. There have been two attempts:
https://github.com/openwrt/packages/pull/9406
https://github.com/openwrt/packages/pull/9408

The latter (from me) was meant as a proof-of-concept for merging _with_ history, which has been rejected as stated earlier. Since I'm not a package maintainer, and my contribution was limited to the history stuff, I will close that one again.

Nevertheless, the terms should be clear now:
_If the routing is supposed to be merged into openwrt/packages, there has to be a PR for every package to be added, and there would be some work to update Makefiles and other stuff.
I'd thus want to encourage package maintainers or other volunteers to start with that (we do not have to transfer everything at once, it would be okay to start with 5 packages) or decide to re-vote and cancel the merge._

zorun, 2020-07-30:
It would be good to reach a decision (one way or the other) before the next 20.XX release.

If I sum up the discussion so far:

## General proposal

Move all packages from openwrt-routing/packages into openwrt/packages

## Arguments in favour

- less overhead, unified organization
- improve package quality thanks to Continuous Integration which is set up in openwrt/packages

## Arguments against

- there is a lot of activity in openwrt/packages, it will be hard to follow what happens to specific routing packages
- more people will be able to merge changes without the approval from maintainer/upstream. This has already been a problem in openwrt-routing (quoting @lindnermarek "a well-intended batman-adv patch pushed by a non-maintainer broke our package while we maintainers were unaware of said change until people complained about the broken package"). It is expected to be a bigger problem in openwrt/packages
- less flexibility (e.g. we lose the possibility to use trunk routing packages on stable openwrt release), although I'm not sure it is actually possible currently.

## Remaining questions

### How to ensure that openwrt-routing maintainers are kept in the loop

This touches both on following activity and ensuring proper patch review.

From the proposed solutions:

- protected branches won't help
- [codeowners file](https://help.github.com/en/articles/about-code-owners) could be used but it needs all maintainers to have commit rights. Also, it seems github-specific?

### Commit rights

Should all current members of openwrt-routing obtain commit rights on openwrt/packages?

### How to import packages

One approach is to move packages individually, improving them in the process. Pro: it improves package quality and consistency. Cons: it takes time and efforts.

The alternative is to bulk-import all packages as they are currently in openwrt-routing.

zorun, 2020-07-30:
From my end, I also think that the activity on openwrt/packages is really too high to follow.  I am subscribed to email notifications for it but I can't keep up with everything.

The advantage of openwrt-routing is that it's a small group of people that (mostly) know each other, so it's easier to work together and trust each other.

If there is a process to make sure that maintainers are made aware of changes and can review patches, then I'm ok with the merge. Otherwise I think it should stay separate, unless there are stronger arguments in favour of the merge.

zorun, 2020-07-30:
Based on feedback from IRC, I have updated the summary (about codeowners files, and importing packages individually)

mwarning, 2020-08-01:
- I do not have a opinion on this
- it would be nice to have a  generic repo in feed.conf.default (commented) that is less professional, thus lowering the bar for participation, but also not being shut out

adrianschmutzler, 2020-08-11:
Looks to me like this will stay separate. Maybe just get CI running and be happy?

aparcar, 2020-08-11:
There is still some discussion at https://github.com/openwrt/packages/issues/12977 to fix the notification issue. Please don't close this just yet.

PolynomialDivision, 2021-02-07:
Can we maybe just adopt the `.github` as workaround? It checks for many targets, so u will notice for example build errors on targets that build with glibc.

aparcar, 2021-03-10:
Hi, it's been some time and I'd like to move this at least over to the openwrt organization so all core members have access. GitHub allows to transfer ownership and renames with automatic forwarding, buildbots internally use the mirror on git.openwrt.org anyway.

After the move I'd make sure that a group called *routing-maintainers* has commit access to the repository and I add all current commiters of openwrt-routing/packages to this group.

Any objections?

lindnermarek, 2021-03-10:
What are all the things which need updating / changing in order to not break after this transition ? URLs, notifications, git repositories, permissions, etc ? A comprehensive list would be appreciated to allow us to catch most of the issues beforehand.

aparcar, 2021-04-08:
GitHub automagically redirects requests meaning we don't have to update anything during the migration, pulling from the old URL will still work. From my understanding the migration also transfers *stars* and notification settings.

Long term the mirror script at git.openwrt.org should be updated.

Apart from that the openwrt.git/README.md needs an update, but that's minor.

BKPepe, 2025-06-17:
I've started working to improve the state of this feed, which has been neglected for a long time. I **agree** that we need to integrate the routing feed into the main packages, but I’m **strongly against**:

- granting commit rights to everyone,
- bulk-merging every package without proper review.

Currently, all OpenWrt feeds (including the main repo) lack active volunteers who consistently:

- review PRs,
- follow the contribution guidelines,
- maintain packages long-term.

There are many packages that were added years ago but are now unmaintained, unused, or stalled in development.

> **Note**: Some packages like `cjdns` **need to be moved** to the packages feed, as the latest version is written in Rust.

---

## Next Steps

- **Bulk merging is a no-go.**
- **Pull Requests will be opened one by one.**

---

## ✅ Packages Ready (once they will be ready once opened PRs will be merged)

| Package Name   | Maintainer(s)                         | Status |
|----------------|----------------------------------------|--------|
| `bird2`        | @tohojo, @PolynomialDivision           | ✅     |
| `bird3`        | @tohojo, @PolynomialDivision           | ✅     |
| `opennds`      | @bluewavenet                           | ✅     |
| `bluewavenet`  | @bluewavenet                           | ✅     |
| `babeld`       | @PolynomialDivision                    | ✅     |
| `alfred`       | @ecsv                                  | ✅     |
| `batcl`        | @ecsv                                  | ✅     |
| `batman-adv`   | @ecsv                                  | ✅     |
| `bmx7`         | @axn                                   | ✅     |
| `nodogsplash`  | @mwarning                              | ✅     |
| `olsrd`        | @PolynomialDivision                    | ✅     |

---

## ❌ Packages with Stalled or Unclear Status

| Package Name     | Maintainer(s)                      | Notes                                                                 | Status |
|------------------|-------------------------------------|-----------------------------------------------------------------------|--------|
| `ahcpd`          | @kerneis                            | Added in 2013, no updates since                                       | ❌     |
| `babel-pinger`   | @tohojo, @PolynomialDivision         | Added in 2014, no updates since                                       | ❌     |
| `batmand`        | @elektra42, @ecsv                   | Last tag 15 years ago, last commit 3 years ago                        | ❌     |
| `naywatch`       | @PolynomialDivision                 | OpenWrt-specific, unclear usage status                                | ❌     |
| `ohybridproxy`   | @sbyx                               | Maintainer inactive, last commit 5 years ago                          | ❌     |
| `oonf-*`         | @Hrogge                             | Not maintained; not building for over a month ([issue #1120](https://github.com/openwrt/routing/issues/1120)) | ❌     |
| `pimbd`          | —                                   | Last upstream commit 7 years ago                                      | ❌     |
| `prince`         | @gabri97                            | Last upstream commit 7 years ago                                      | ❌     |
| `vis`            | @elektra42                          | Severely outdated; last upstream commit in 2016                       | ❌     |

---

## 🔧 Package Requiring Refactoring

| Package Name | Maintainer | Notes                                 | Status |
|--------------|------------|----------------------------------------|--------|
| `ndppd`      | @Red54     | Needs to be refactored before review   | ❌     |

I hope I didnt miss any other packages and sometimes, it is just mess. LuCI packages are in separate folder or even included in the package Makefile.

dangowrt, 2025-06-17:
From the actively maintained packages in the routing feed it seems to me that only @bluewavenet and @ecsv. I'm all for giving commit access to those two once the packages they maintained up to now are merged in the packages feed.

PolynomialDivision, 2025-06-19:
naywatch is still in use.

bluewavenet, 2025-06-19:
@BKPepe 

This:
> opennds 	@bluewavenet 	✅
> bluewavenet 	@bluewavenet 	✅

should read:
opennds 	@bluewavenet 	✅
mesh11sd 	@bluewavenet 	✅

wfleurant, 2025-06-20:
cjdns packages aught to be included.. if put into main packages then the potential to bump the version to the rust enhanced version is possible (with help).

fingon, 2025-06-20:
Ohybridproxy was primarily used with hnetd which is gone. So I suspect it will not be missed much ( I used it standalone in some network too but I doubt almost anyone else ever has ).

geldot, 2026-08-16:
> Alright guys,
> 
> This issue has been around for 10 years, and we just keep coming back to it. Even I am returning to it after a year, but I was hoping someone might be a bit more proactive especially since there was a table showing what's ready and just grab their own package and move it.
> 
> That didn't happen. 10 years is a long time, so let's finally get this sorted out. :) The pull requests have been created.

Please consider merging this repository's history into `packages` with a merge commit (`--allow-unrelated-histories`), and avoid making unrelated changes during the migration to maintain clear provenance.