openwrt/routing #210: batman-adv: Build error due to wrong include order

yanosz, 2016-07-29:
Building batman-adv (branch: for-15.05, using the 15.05.1 SDK) I got:
`OpenWrt-SDK-15.05.1-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/build_dir/target-mips_34kc_uClibc-0.9.33.2/batman-adv/batman-adv-2016.1/net/batman-adv/types.h:388:26: error: expected ')' before numeric constant
 DECLARE_EWMA(throughput, 1024, 8)`

This is fixed in: https://patchwork.open-mesh.org/patch/16488/ - please apply this patch

lindnermarek, 2016-07-30:
@yanosz: As I wrote on IRC: Did you test the patch and verify this fixes the issue ?
it wasn't applied to the routing feed yet because we thought only the master branch needed it due to newly added code.

yanosz, 2016-07-30:
Hmm... this is strange. I doesn't work altough I remembered it to work yesterday :-(

yanosz, 2016-07-30:
As stated in IRC this might be a kernel version issue - 

[10:18:19] <yanosz_> ... maybe its an kernel version issue .... 
[10:18:44] <yanosz_> DECLARE_EWMA was not in average.h before 4.3
[10:18:46] <yanosz_> http://lxr.free-electrons.com/source/include/linux/average.h?v=4.2
[10:18:59] <yanosz_> ... just guessing
[10:24:52] <yanosz_> thinks seem to be strange ... after adding -I$(PKG_BUILD_DIR)/compat-include/ \ manually I get another error: /home/jan/src/routing_pages/OpenWrt-SDK-15.05.1-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/build_dir/target-mips_34kc_uClibc-0.9.33.2/batman-adv/batman-adv-2016.1/compat-hacks.h:58:1: error: conflicting types for 'batadv_ethtool_get_link_ksettings'
[10:24:52] <yanosz_>  manu
[10:26:04] <yanosz_> I decided to do so, since DECLARE_EWMA is present here... somebody seem to see a need for backporting this kernel macro to openwrt ... but it is not included in the makefile
[10:26:14] <yanosz_> by default

lindnermarek, 2016-07-31:
@yanosz: Could you give this patch a try: marek/batman-adv/for-15.05-ewma-compat ?

I added the missing ewma compat to the batman-adv compat file which should resolve your issue without having to tamper with any includes.

yanosz, 2016-08-01:
Thanks .. but there's another issue:

```
c-0.9.33.2/batman-adv/batman-adv-2016.1/net/batman-adv/fragmentation.o
  CC [M]  /home/jan/src/build_openwrtrouting/OpenWrt-SDK-15.05.1-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/build_dir/target-mips_34kc_uClibc-0.9.33.2/batman-adv/batman-adv-2016.1/net/batman-adv/gateway_client.o
/home/jan/src/build_openwrtrouting/OpenWrt-SDK-15.05.1-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/build_dir/target-mips_34kc_uClibc-0.9.33.2/batman-adv/batman-adv-2016.1/net/batman-adv/gateway_client.c: In function 'batadv_write_buffer_text':
/home/jan/src/build_openwrtrouting/OpenWrt-SDK-15.05.1-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/build_dir/target-mips_34kc_uClibc-0.9.33.2/batman-adv/batman-adv-2016.1/net/batman-adv/gateway_client.c:620:2: error: implicit declaration of function 'seq_has_overflowed' [-Werror=implicit-function-declaration]
  ret = seq_has_overflowed(seq) ? -1 : 0;
```

lindnermarek, 2016-08-03:
@yanosz: Can you provide some information how to replicate this issue with 15.05 ? Might be easier to move forward if I could see the issues myself. I am wondering why all this isn't working in your setup ...

yanosz, 2016-08-04:
I do this:
wget https://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/OpenWrt-SDK-15.05.1-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64.tar.bz2
tar xjf OpenWrt-SDK-15.05.1-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64.tar.bz2
cd OpenWrt-SDK-15.05.1-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/
git clone https://github.com/openwrt-routing/packages.git -b marek/batman-adv/for-15.05-ewma-compat
cd package
ln -s ../packages/batman-adv/
ln -s ../packages/batctl/
cd ..
make world -j1 V=99

Error appears

nzamps, 2016-08-15:
I came across this today, and I had to create a patch in target/linux/generic/patches-3.18:

`--- a/include/linux/seq_file.h	2016-08-16 09:51:48.293174604 +1200
+++ b/include/linux/seq_file.h	2016-08-16 09:52:45.169174234 +1200
@@ -43,6 +43,21 @@
 #define SEQ_SKIP 1
 
 /**
+ * seq_has_overflowed - check if the buffer has overflowed
+ * @m: the seq_file handle
+ *
+ * seq_files have a buffer which may overflow. When this happens a larger
+ * buffer is reallocated and all the data will be printed again.
+ * The overflow state is true when m->count == m->size.
+ *
+ * Returns true if the buffer received more than it can hold.
+ */
+static inline bool seq_has_overflowed(struct seq_file *m)
+{
+        return m->count == m->size;
+}
+ 
+/**
  * seq_get_buf - get buffer to write arbitrary data to
  * @m: the seq_file handle
  * @bufp: the beginning of the buffer is stored here

`

ecsv, 2016-09-02:
The batman-adv package in the git branch for-15.05 is created for the OpenWrt Chaos Calmer git branch. It works fine in this combination.

When you manually hack around with the SDK then please make sure that it contains the correct mac80211/backports files from the current OpenWrt Chaos Calmer git version.  The SDK link you gave us contains an older version which doesn't have the correct backports files. You can check this by doing a simple grep:

    $ cd OpenWrt-SDK-15.05.1-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64
    $ grep -rli seq_has_overflowed ./staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include/mac80211*
   $ cd ../openwrt_cc
   $ grep -rli seq_has_overflowed ./staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include/mac80211*
   ./staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include/mac80211-backport/linux/seq_file.h

So you can ask OpenWrt to update their SDK with a new Chaos Calmer release. It is required anyway because Chaos Calmer doesn't build at the moment on systems with GCC6 (see https://github.com/openwrt/openwrt/pull/83)

Btw. the patch you mentioned (https://patchwork.open-mesh.org/patch/16488/) will not work here. It patches a file which is not used by OpenWrt at all.

Bastien-, 2016-09-06:
I'm having the same issue. I'm not sure to understand the fix. Do you recommend a workaround?

ecsv, 2016-09-07:
Please don't use the OpenWrt SDK together with the OpenWrt *routing* Chaos Calmer development branch/repository. Someone from from OpenWrt has to make a new SDK (15.05.2?) based on the Chaos Calmer development branch/repository before this works again.

Try to use the OpenWrt Chaos calmer development branch/repository together with the OpenWrt *routing* Chaos Calmer branch/repository. This means: build everything from source

bk138, 2016-11-17:
FWIW, I was able to build 2016.1 with the 15.05 SDK. In fact, I had to, since I want to leave the precompiled, shipped kernel as-is. The batman_adv kernel module I built from the git buildroot (branch chaos_calmer) had a different kernel dependeny (kernel (= 3.18.20-1-65d6d011e5f83c59832c63b917f2ea09) then the 15.05 kernel ( kernel (=3.18.20-1-7bed08fa9c06eb8089e82c200340ec66), but could be made to install via opkg --force-depends. But in the end it would throw 
```
Mon Mar  7 15:49:41 2016 kern.warn kernel: [   35.910000] Unhandled kernel unaligned access[#1]:
Mon Mar  7 15:49:41 2016 kern.warn kernel: [   35.910000] CPU: 0 PID: 393 Comm: kworker/u2:3 Not tainted 3.18.20 #1
Mon Mar  7 15:49:41 2016 kern.warn kernel: [   35.910000] Workqueue: bat_events 0x86af1ec4 [batman_adv@86ae0000+0x1d1d0]
# regdump and call trace and whatever follow...
```
Long story short, here's what I did:

* downloaded https://downloads.openwrt.org/chaos_calmer/15.05/ar71xx/generic/OpenWrt-SDK-15.05-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64.tar.bz2 and unpacked
* in there:
```
make menuconfig
make
scripts/feeds update # contain batman-adv 2016.1 atm
scripts/feeds install kmod-batman-adv
scripts/feeds install batctl
make menuconfig # to check that batman stuff was enabled but it already was
```
Now this wouldn't build cause of issues mentioned. Thus, I added those two snippets to package/feeds/routing/batman-adv/files/compat-hacks.h:

```
#if LINUX_VERSION_CODE < KERNEL_VERSION(4, 3, 0)

/* Exponentially weighted moving average (EWMA) */

#define DECLARE_EWMA(name, _factor, _weight)                            \
        struct ewma_##name {                                            \
                unsigned long internal;                                 \
        };                                                              \
        static inline void ewma_##name##_init(struct ewma_##name *e)    \
        {                                                               \
                BUILD_BUG_ON(!__builtin_constant_p(_factor));           \
                BUILD_BUG_ON(!__builtin_constant_p(_weight));           \
                BUILD_BUG_ON_NOT_POWER_OF_2(_factor);                   \
                BUILD_BUG_ON_NOT_POWER_OF_2(_weight);                   \
                e->internal = 0;                                        \
        }                                                               \
        static inline unsigned long                                     \
        ewma_##name##_read(struct ewma_##name *e)                       \
        {                                                               \
                BUILD_BUG_ON(!__builtin_constant_p(_factor));           \
                BUILD_BUG_ON(!__builtin_constant_p(_weight));           \
                BUILD_BUG_ON_NOT_POWER_OF_2(_factor);                   \
                BUILD_BUG_ON_NOT_POWER_OF_2(_weight);                   \
                return e->internal >> ilog2(_factor);                   \
        }                                                               \
        static inline void ewma_##name##_add(struct ewma_##name *e,     \
                                             unsigned long val)         \
        {                                                               \
                unsigned long internal = ACCESS_ONCE(e->internal);      \
                unsigned long weight = ilog2(_weight);                  \
                unsigned long factor = ilog2(_factor);                  \
                                                                        \
                BUILD_BUG_ON(!__builtin_constant_p(_factor));           \
                BUILD_BUG_ON(!__builtin_constant_p(_weight));           \
                BUILD_BUG_ON_NOT_POWER_OF_2(_factor);                   \
                BUILD_BUG_ON_NOT_POWER_OF_2(_weight);                   \
                                                                        \
                ACCESS_ONCE(e->internal) = internal ?                   \
                        (((internal << weight) - internal) +            \
                                (val << factor)) >> weight :            \
                        (val << factor);                                \
        }

#endif /* < KERNEL_VERSION(4, 3, 0) */
```
and:
```
#if LINUX_VERSION_CODE < KERNEL_VERSION(3, 19, 0)

static inline bool seq_has_overflowed(struct seq_file *m)
{                       
        return m->count == m->size;
}                       
        
#endif /* < KERNEL_VERSION(3, 19, 0) */
```
Both can be found in batman-adv's source package's compat-include/ dir.
With those added, I could build the kernel module as well as batctl for 15.05 box. Running fine so far.

ecsv, 2019-05-24:
Building the openwrt-routing master branch against chaos calmer is not supported. Issue should be closed.

closed 2019-05-24 by simonwunderlich