openwrt/routing #211: babeld:  error is: Package babeld is missing dependencies for the following libraries

yanosz, 2016-07-29:
Trying to build babeld (branch: for-15.05, using SDK version 15.05.01) it failed:

`install -m0755 ./files/babeld.init /home/jan/src/routing_pages/OpenWrt-SDK-15.05.1-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/build_dir/target-mips_34kc_uClibc-0.9.33.2/babeld-1.7.1/ipkg-ar71xx/babeld/etc/init.d/babeld
find /home/jan/src/routing_pages/OpenWrt-SDK-15.05.1-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/build_dir/target-mips_34kc_uClibc-0.9.33.2/babeld-1.7.1/ipkg-ar71xx/babeld -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package babeld is missing dependencies for the following libraries:
libc.so.0
libgcc_s.so.1
Makefile:61: recipe for target '/home/jan/src/routing_pages/OpenWrt-SDK-15.05.1-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/bin/ar71xx/packages/base/babeld_1.7.1-1_ar71xx.ipk' failed
make[3]: *** [/home/jan/src/routing_pages/OpenWrt-SDK-15.05.1-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/bin/ar71xx/packages/base/babeld_1.7.1-1_ar71xx.ipk] Error 1
make[3]: Leaving directory '/home/jan/src/routing_pages/packages/babeld'
package/Makefile:191: recipe for target 'package/babeld/compile' failed
make[2]: *** [package/babeld/compile] Error 2
make[2]: Leaving directory '/home/jan/src/routing_pages/OpenWrt-SDK-15.05.1-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64'
package/Makefile:188: recipe for target '/home/jan/src/routing_pages/OpenWrt-SDK-15.05.1-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_compile' failed
make[1]: *** [/home/jan/src/routing_pages/OpenWrt-SDK-15.05.1-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jan/src/routing_pages/OpenWrt-SDK-15.05.1-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64'
/home/jan/src/routing_pages/OpenWrt-SDK-15.05.1-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/include/toplevel.mk:174: recipe for target 'world' failed
make: *** [world] Error 2`

kerneis, 2016-07-29:
@zorun any idea what is going on? I don't remember when I last tried to build 15.05.

yanosz, 2016-07-30:
Hmm... this is strange. It doesn't seem to work, although I rember it to work yesterday :-/

zorun, 2016-07-30:
I never used the SDK, I have no idea how it works.

I'll try building babeld from the latest OpenWRT 15.05 git.

zorun, 2016-07-30:
`babeld` builds fine on the CC branch of OpenWRT (chaos_calmer@49403).

closed 2020-01-11 by zorun