openwrt/routing #216: Alfred: Build fails due to http-403

yanosz, 2016-07-30:
See: https://s3.amazonaws.com/archive.travis-ci.org/jobs/148521584/log.txt

`make[3]: Entering directory `/home/travis/build/yanosz/packages-1/alfred'
mkdir -p /home/travis/build/yanosz/packages-1/tests/OpenWrt-SDK-15.05.1-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/dl
/home/travis/build/yanosz/packages-1/tests/OpenWrt-SDK-15.05.1-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/scripts/download.pl "/home/travis/build/yanosz/packages-1/tests/OpenWrt-SDK-15.05.1-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/dl" "alfred-2014.4.0.tar.gz" "053cb5d9e7ca9384598e82944343dea2" "http://downloads.open-mesh.org/batman/releases/batman-adv-2014.4.0"
--2016-07-30 11:48:45--  http://downloads.open-mesh.org/batman/releases/batman-adv-2014.4.0/alfred-2014.4.0.tar.gz
Resolving downloads.open-mesh.org (downloads.open-mesh.org)... 78.46.248.236, 2a01:4f8:141:3341:78:46:248:236
Connecting to downloads.open-mesh.org (downloads.open-mesh.org)|78.46.248.236|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://downloads.open-mesh.org/batman/releases/batman-adv-2014.4.0/alfred-2014.4.0.tar.gz [following]
--2016-07-30 11:48:45--  https://downloads.open-mesh.org/batman/releases/batman-adv-2014.4.0/alfred-2014.4.0.tar.gz
Connecting to downloads.open-mesh.org (downloads.open-mesh.org)|78.46.248.236|:443... connected.
HTTP request sent, awaiting response... 403 Forbidden
2016-07-30 11:48:46 ERROR 403: Forbidden.

Download failed.
--2016-07-30 11:48:46--  http://mirror2.openwrt.org/sources/alfred-2014.4.0.tar.gz
Resolving mirror2.openwrt.org (mirror2.openwrt.org)... 46.4.11.11
Connecting to mirror2.openwrt.org (mirror2.openwrt.org)|46.4.11.11|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-07-30 11:48:47 ERROR 404: Not Found.

Download failed.
--2016-07-30 11:48:47--  http://downloads.openwrt.org/sources/alfred-2014.4.0.tar.gz
Resolving downloads.openwrt.org (downloads.openwrt.org)... 78.24.191.177
Connecting to downloads.openwrt.org (downloads.openwrt.org)|78.24.191.177|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-07-30 11:48:48 ERROR 404: Not Found.

Download failed.
No more mirrors to try - giving up.`

FreifunkUFO, 2016-08-03:
maybe you dont have internet? downloading that mentioned file is working for me:

https://downloads.open-mesh.org/batman/releases/batman-adv-2014.4.0/alfred-2014.4.0.tar.gz

yanosz, 2016-08-04:
Travis-CI has internet - for sure.
Point is: The download is rejected by the server (http-403)

ecsv, 2016-09-02:
Please use a wget which supports SNI. See https://bugs.launchpad.net/ubuntu/+source/wget/+bug/1580700. This should for example be fixed by by upgrading your Travis CI to Trusty as explained in https://docs.travis-ci.com/user/trusty-ci-environment/

ecsv, 2019-05-24:
This is a problem on the test computer (fixable by an update of wget). The ubuntu version used in the test is no longer supported. The issue should be closed

closed 2019-05-24 by simonwunderlich