openwrt/routing #240: $gatewaymac, $clientmac not available

metsis, 2016-11-11:
although documentation states that $gatewaymac and $clientmac should be available on the splash page through nodogpsplash preprocessor, i am not getting them. have tested the same with e.g. $uptime and $tok, $loc, and they work without problems. any ideas what might cause this? 

i am running version 0.9_beta9.9.9 - might this be too old?

lynxis, 2016-11-27:
@metsis you might want to move this bug to the nodogsplash project (also on github). @mwarning looks like this is your bug ;).
.

mwarning, 2016-11-27:
@metsis $gatewaymac/$clientmac was implemented in v1.0.0. The changelog had a duplicate entry. It is fixed now.
Please open the ticket if the problem is still in v1.0.0.

closed 2016-11-27 by mwarning