openwrt/routing #244: olsrd build broken since commit 3b14a1c5b6aaf0422374a14998279d9047b0abf7

dangowrt, 2016-12-17:
buildbot says
```
make[6]: Entering directory '/home/lede/buildbot/mips_24kc/build/sdk/build_dir/target-mips_24kc_musl-1.1.15/olsrd-0.9.5/lib/pud/nmealib'
[CC] src/sentence.c
cc1: error: /home/lede/buildbot/mips_24kc/build/sdk/staging_dir/target-mips_24kc_musl-1.1.15/include: No such file or directory [-Werror=missing-include-dirs]
cc1: all warnings being treated as errors
Makefile:57: recipe for target 'build/sentence.o' failed
make[6]: *** [build/sentence.o] Error 1
make[6]: Leaving directory '/home/lede/buildbot/mips_24kc/build/sdk/build_dir/target-mips_24kc_musl-1.1.15/olsrd-0.9.5/lib/pud/nmealib'
Makefile:162: recipe for target 'nmealib' failed
make[5]: *** [nmealib] Error 2
make[5]: Leaving directory '/home/lede/buildbot/mips_24kc/build/sdk/build_dir/target-mips_24kc_musl-1.1.15/olsrd-0.9.5/lib/pud'
Makefile:241: recipe for target 'libs' failed
make[4]: *** [libs] Error 2
```

dangowrt, 2016-12-17:
introduced by commit 3b14a1c5b6aaf0422374a14998279d9047b0abf7
@bittorf

bittorf, 2016-12-18:
thank you: hopefully fixed with https://github.com/openwrt-routing/packages/commit/dcecd7007d0557a52ad619e3ef9815417ae85349

SvenRoederer, 2016-12-21:
dcecd70 did not fix

Just checked with LEDE + OpenWrt SDK (```feeds install olsrd; make```)

bittorf, 2016-12-21:
@SvenRoederer - do you have an idea what to change?

bittorf, 2016-12-21:
@SvenRoederer does it work for you, if you explicitely select 'gpsd' in menuconfig?

SvenRoederer, 2016-12-22:
selecting "gpsd" explicitly buiilds it correctly before building olsrd.
But building olsrd still fails like before

SvenRoederer, 2016-12-22:
This error is as of a missing includedir:
```make[6]: Entering directory '/tmp/lede-sdk-ramips-mt7621_gcc-5.4.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kc_musl-1.1.15/olsrd-0.9.5/lib/pud/nmealib'
[CC] src/sentence.c
cc1: error: /tmp/lede-sdk-ramips-mt7621_gcc-5.4.0_musl-1.1.15.Linux-x86_64/staging_dir/target-mipsel_24kc_musl-1.1.15/include: No such file or directory [-Werror=missing-include-dirs]
cc1: all warnings being treated as errors
Makefile:57: recipe for target 'build/sentence.o' failed
make[6]: *** [build/sentence.o] Error 1
make[6]: Leaving directory '/tmp/lede-sdk-ramips-mt7621_gcc-5.4.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kc_musl-1.1.15/olsrd-0.9.5/lib/pud/nmealib'
```
Checked the environment for any references, as the path is not found in the Makefile explicitly and found "CXXFLAGS" causing this: '-Os -pipe ....  -I/tmp/lede-sdk-ramips-mt7621_gcc-5.4.0_musl-1.1.15.Linux-x86_64/staging_dir/target-mipsel_24kc_musl-1.1.15/include  ...'

removing "-Wmissing-include-dirs" in https://github.com/OLSR/olsrd/blob/master/lib/pud/nmealib/Makefile.inc#L66 disables this warning here and fixes build.
Maybe a patch in this package is an option ...

FreifunkUFO, 2016-12-30:
see also https://github.com/openwrt-routing/packages/pull/242
@dangowrt is the problem still existing? is there an URL for the buildbot?

PR #246 (2017-01-04): olsrd: disable compiler-warning "missing-include-dirs"
remove this warning as it fails to build for the nmealib with missing
"staging_dir/target-mipsel_24kc_musl-1.1.15/include" from buildenv

this fixes #244 on OpenWrt + LEDE master (checked via SDK-build)
closed 2017-01-04 by bittorf