openwrt/routing #251: olsrd: master is missign dep to libcrypto.so.1.0.0

SvenRoederer, 2017-01-09:
olsrd requires libcrypto by default since https://github.com/OLSR/olsrd/commit/1130cbf9146502d56c135081af7f5d1a5b605d09.

Will we need this feature (this will required dependency to openssl)? see https://github.com/OLSR/olsrd/issues/13

bittorf, 2017-01-10:
IMHO this feature makes sense. openssl-dev is only needed on the buildhost - isn't it?

jow-, 2017-01-10:
I *don't* think it makes sense to require OpenSSL on the target. This library is huge, prone to security issues and frequently requires security updates. It is everything you usually do *not* want on an embedded mesh node.

bittorf, 2017-01-10:
@jow: again: i wrote: "openssl-dev is only needed on the buildhost - isn't it?"

jow-, 2017-01-10:
From looking at the commit it looks like the olsrd binary also links libcrypto.so on the target, so no, it is not needed on the host but required as full runtime library on the target.

It also seems like one needs `MAKE_VARS += NOCONFIGCHECKSUM=1` or something to inhibit linking. Passing the define is not enough.

jow-, 2017-01-10:
To further elaborate - the linked commit does not explain why the checksumming was added but I guess it is to track configuration file updates. The `procd` daemon on LEDE/OpenWrt already handles tracking config file updates and command line changes so it definitely makes no sense to have olsrd do the same so *imho* it makes sense to disable olsrd's own checksumming feature here.

bittorf, 2017-01-10:
@fhuberts: i try to understand why this commit was added and wonder why i cant see it with 'git log'. what works is:
```
git show 1130cbf9146502d56c135081af7f5d1a5b605d09
and
git branch --contains 1130cbf9146502d56c135081af7f5d1a5b605d09
(outputs 'master')
```
what is not working:
```
git log
(it simply does not show this - and the surrounding - commits - why?)
```

fhuberts, 2017-01-10:
As jow said, you can just disable the feature with that define.
I'll see it the licenses are compatible enough to include only the needed code of openssl into the olsrd tree.

fhuberts, 2017-01-10:
Explanation is here: https://github.com/OLSR/olsrd/commit/1f976fa1f56e38273fbf5589ade2a42e7a083494

bittorf, 2017-01-10:
@fhuberts can we please use an internal hash-function for that? something like http://www.azillionmonkeys.com/qed/hash.html - ("SuperFastHash") - or one of the others... ferry really: calling openssl for that seems overkill...

@jow-  this hash is not about internal reloading of config-changes, but the reflect a changed config via the json-output...

fhuberts, 2017-01-10:
@jow- @bittorf  olsrd master now has the sha256 checksum code in-tree, no more linking to libcrypto

bittorf, 2017-01-10:
@fhuberts: i'am not happy with the resulting bloat. it adds here (x86) ~28.000 bytes to the olsrd-binary.

fhuberts, 2017-01-10:
Really?

On my system, on a release (non-debug) build, 64 bits x86_64:
* pre: -rwxr-xr-x. 1 ferry ferry 340064 Jan 10 14:45 /home/ferry/vcs/nbs/nbsolsrd/dist/usr/sbin/olsrd
* post: -rwxr-xr-x. 1 ferry ferry 348224 Jan 10 14:46 /home/ferry/vcs/nbs/nbsolsrd/dist/usr/sbin/olsrd

That is a diff of 8160 bytes

On my system, on a release (non-debug) build, 32 bits i386:
* pre: -rwxr-xr-x. 1 ferry ferry 383528 Jan 10 14:49 /home/ferry/vcs/nbs/nbsolsrd/dist/usr/sbin/olsrd
* post: -rwxr-xr-x. 1 ferry ferry 391720 Jan 10 14:48 /home/ferry/vcs/nbs/nbsolsrd/dist/usr/sbin/olsrd

That is a diff of 8192 bytes

Either way, you can't have your cake and eat it too.
It's either linking to libcrypto (~2.8M) or this.

If you want a different hash algo, then submit a PR :-)
I'll merge it if it's good.

Note that the current algo does streaming of the config file into the checksum...

bittorf, 2017-01-10:
@fhuberts - thanks for comparing. i'am OK with the approach now and will submit a PR with another algo if there is a long dark winter-evening - thumbs up!

closed 2017-01-10 by bittorf

jow-, 2017-01-10:
I second @bittorf here, imho SFH hash would be a good candidate here since it is only a few lines of C to implement, is very fast and easy to feed incrementally.

It obviously does not make crypto-relevant guarantees like SHA-256 but if the hash is merely used for detecting changes in the config source then it would work well most likely.

fhuberts, 2017-01-10:
You guys are soooo in luck...

I was curious, and SFH was quite easy to integrate, so I did it and it's now on master.

It's way less accurate but it should be sufficient.

Also, it's about 100 bytes less than when olsrd linked against libcrypto ;-)

bittorf, 2017-01-10:
@fhuberts kisses are going out - thank you!