openwrt/routing #254: Providing a "lede-17.01" branch for the upcoming LEDE release

jow-, 2017-01-15:
Hello @openwrt-routing/owners -

in preparation for the upcoming LEDE 17.01 release I'd like to create a new branch "lede-17.01" in this repository which will be used thorough the lifetime of the LEDE 17.01 release series.

The branch will serve multiple purposes:
- It allows the LEDE release to use a stable (as in not rapidly changing) code base for building packages
- It allows to selectively backport fixes into the existing LEDE release
- It allows package maintainers to focus on their work without having to worry about breaking things

LEDE will continually rebuild the "lede-17.01" feed branch and provide updated binaries thorough the life time of the 17.01 release series.

The new branch shall only receive important bug or security fixes. Major package updates should be avoided.

I am asking for your permission to...
 - create a "lede-17.01" branch in this repository
 - cherry-pick fixes, especially security related ones, from master into said branch with or without the maintainers prior consent

Package maintainers are very welcome, but not required, to help maintaining the stable release branch together with the LEDE team. Branching for LEDE 17.01 is planned on Monday, the 16th and I'd like to have the branch in place until then.

I'd appreciate to receive some short acknowledgements before proceeding here to ensure that there's a consensus among the maintainers of this repository.

Thank you for your great work on the package feeds which turns OpenWrt and LEDE into such versatile distributions! :)

mwarning, 2017-01-15:
> I am asking for your permission to...
>  - create a "lede-17.01" branch in this repository
>  - cherry-pick fixes, especially security related ones, from master into said branch with or without the maintainers prior consent

As for me, I like to say: go ahead. Thanks for your work!

axn, 2017-01-15:
On 15.01.2017 15:28, Moritz Warning wrote:
>> I am asking for your permission to...
>> - create a "lede-17.01" branch in this repository
>> - cherry-pick fixes, especially security related ones, from master
> into said branch with or without the maintainers prior consent
> 
> As for me, I like to say: go ahead. Thanks for your work!
> 

Same for me! Please proceed! And thanks for your work!
/axel

zorun, 2017-01-15:
Good for me too, the whole release plan is quite clear and consistent!

zioproto, 2017-01-16:
OK !

closed 2017-09-14 by zorun