openwrt/routing #272: olsrd: since update to 0.9.6 the plugins config needs to be adapted

thuehn, 2017-02-07:
Since commit: olsrd: update to v0.9.6 - contains important fixes (dec45d4df52b05c0e9cd36d373682559c13e6c59) certain olsrd plugins got updated and use a different version number in the plugin name. The is not auto migration file for your current olsrd.conf, so those updated plugins will not start without adapting the uci settings.

For example, the netjson plugin and hence the neigh.sh script will not work after this olsrd update to 0.9.6. You need to change your /etc/config/olsrd from:

config LoadPlugin
        option library 'olsrd_jsoninfo.so.0.0'
        option accept '0.0.0.0'
        option ignore '0'

to:
config LoadPlugin
        option library 'olsrd_jsoninfo.so.1.1'
        option accept '0.0.0.0'
        option ignore '0'

Greetings Thomas

bittorf, 2017-02-07:
the was already a discussion about that. what is possible, that we extend ```olsrd_write_loadplugin()``` in ```/lib/functions/olsrd.sh``` for autoloading the newest available plugin if they are without version e.g.:

```
# instead of
olsrd.@LoadPlugin[0].library='olsrd_txtinfo.so.1.1'

# we now allow:
olsrd.@LoadPlugin[0].library='olsrd_txtinfo'
# and this autoselects the most recent plugin
```

 does this make sense? or at least show a warning at startup?
auto-changing uci-values is a no-go IMHO. @SvenRoederer what do you think about this approach.

thuehn, 2017-02-07:
Including olsrd plugins by name without version numers would be a migration path. 
+1 from my side, as you still can compile an older plugin version to be used

bittorf, 2017-02-12:
so https://github.com/openwrt-routing/packages/commit/f4bc87dc65729ef663f6dc9b7e157a909c5eb92b

SvenRoederer, 2017-02-12:
@bittorf looks good

bittorf, 2017-02-13:
i think this is solved - thanks for the input - lets focus now on https://github.com/OLSR/olsrd/issues/20

closed 2017-02-13 by bittorf