openwrt/routing #274: luci-app-cjdns: adding UDP-IPv6-Peers to IPv4 connectTo entry

chpio, 2017-02-11:
The app adds the IPv6 peer to the IPv4 section resulting in a crash.

> 2017-02-11T11:54:39+01:00 notice cjdns[]: 1486814079 CRITICAL Configurator.c:110 Got error [different address type than this socket is bound to.] calling [UDPInterface_beginConnection]
2017-02-11T11:54:39+01:00 notice cjdns[]: 1486814079 CRITICAL Configurator.c:60 enable Log_LEVEL=KEYS to see message content.
2017-02-11T11:54:39+01:00 notice cjdns[]: 1486814079 CRITICAL Configurator.c:74 Aborting.

expected: 
```json
"UDPInterface":[
    {
      "bind":"0.0.0.0:123456",
      "connectTo":{
        "[2a01::1234:4321]:123456":{
          "publicKey":"xxx",
          "password":"xxx",
          "user":"user"
        }
      }
    },
    {
      "bind":"[::]:27062",
      "connectTo":[]
    }
]
```

actual output:
 ```json
"UDPInterface":[
    {
      "bind":"0.0.0.0:123456",
      "connectTo":[]
    },
    {
      "bind":"[::]:27062",
      "connectTo":{
        "[2a01::1234:4321]:123456":{
          "publicKey":"xxx",
          "password":"xxx",
          "user":"user"
        }
      }
    }
],
```

chpio, 2017-02-22:
It's actually the misleading options listed at *Peers* > *Outgoing UDP Peers* > *UDP interface*. The IPv4 UDP interface is labelled as "[::]:$port", which is definitely wrong. The correct labelling for the options should be: "$addr:$port" ($addr and $port coming from the entries at *Settings* > *UDP Interfaces*).

ps: If i set that to *custom* and then type *2* the entry is added to the ipv6 section.

chpio, 2017-02-22:
I think (im not a lua dev, so im maybe wrong) this looks weird:
https://github.com/openwrt-routing/packages/blob/89914e47de14d11aaffd5472a3b44c0130e1869f/luci-app-cjdns/luasrc/model/cbi/cjdns/peering.lua#L42-L45

```lua
udp_interface:value(index, section.address .. ":" .. section.port)
```
should be:
```lua
udp_interface:value(i, section.address .. ":" .. section.port)
```

lgierth, 2017-02-22:
Nice find, I think you're right. I wonder why nobody ever noticed this form field is broken. Wanna give try fixing it? Bonus points for testing it on a device :)

lgierth, 2017-02-22:
(Assigning myself just so I don't forget having another look at this soon-ish.)

closed 2019-01-09 by chpio