openwrt/routing #277: Remove or keep bird-openwrt packets?

p4u, 2017-02-21:
The bird integration to UCI/LUCI was made by Eloi, a student which is now out of the map. I pushed the packets to the oprnwrt-routing repository with the idea that he would maintain them, however I don't think it is gonna happen. Is there anyone interested on takin care of the maintenance? If now I would remove them at least from the official repository.  

https://github.com/openwrt-routing/packages/tree/master/bird-openwrt

eloicaso, 2017-02-21:
Dear Pau,

As you say, I have not been actively maintaining the package (lack of time). However, I just have started updating it as I could make it part of my MSc. project.

tohojo, 2018-04-03:
Any plans to support Bird 2.0?

eloicaso, 2018-04-03:
Yes, I already started looking at it and how Bird2 manages configuration now. However, I have had no spare time yet to properly start developing it. Moreover, I am still unsure about how bird1/2 separation will be handled (https://github.com/openwrt-routing/packages/pull/340)

tohojo, 2018-04-03:
Eloi <notifications@github.com> writes:

> Yes, I already started looking at it and how Bird2 manages
> configuration now. However, I have had no spare time yet to properly
> start developing it. Moreover, I am still unsure about how bird1/2
> separation will be handled
> (https://github.com/openwrt-routing/packages/pull/340)

Ah, awesome, didn't see that pull request. Not sure there's a reason to
keep bird 1.6 around if you can support the 2.0 config format in uci.
The upgrade cost is mostly in configuration file format changes...

briantopping, 2020-03-13:
Please don't delete this package. Even if it is out of date, it is better than nothing.

rogerpueyo, 2023-03-06:
Hi @p4u,

I'm maintaining the bird1-openwrt packages now, so you may want to close this issue.

Cheers!

closed 2023-03-06 by BKPepe