openwrt/routing #291: upnp not opens ports for aMule

BUG-reporter, 2017-03-21:
Hello. I'm using Mikrotik RB941-2nD (hAP lite) with LEDE trunk (now it's SNAPSHOT, r32-0685f2a)
 for a several months upgrading from time to time to the latest trunk. The thing is for some reason miniupnp not opens ports for aMule. All the time i'm getting lowid. When I'm starting aMule nothing happens and Luci uPnP page still shows "There are no active redirects". Also all I found in logs is the following:
> Tue Mar 21 03:11:00 2017 daemon.err miniupnpd[5019]: could not open lease file: /var/upnp.leases
> Tue Mar 21 03:11:00 2017 daemon.notice miniupnpd[5019]: HTTP listening on port 5000
> Tue Mar 21 03:11:00 2017 daemon.notice miniupnpd[5019]: HTTP IPv6 address given to control points : [fdc1:fc19:7e00::1]
> Tue Mar 21 03:11:00 2017 daemon.notice miniupnpd[5019]: Listening for NAT-PMP/PCP traffic on port 5351
Also here is my config:
```
/etc/config# cat upnpd 

config upnpd 'config'
        option download '1024'
        option upload '512'
        option internal_iface 'lan'
        option port '5000'
        option upnp_lease_file '/var/upnp.leases'
        option enabled '1'
        option uuid 'a5139dc8-a381-4202-8a37-2db971d0da05'
        option log_output '1'

config perm_rule
        option action 'allow'
        option ext_ports '1024-65535'
        option int_addr '0.0.0.0/0'
        option int_ports '1024-65535'
        option comment 'Allow high ports'

config perm_rule
        option action 'deny'
        option ext_ports '0-65535'
        option int_addr '0.0.0.0/0'
        option int_ports '0-65535'
        option comment 'Default deny'
```
The thing is I have also another router (TP-Link) with OpenWRT latest release and there all is working OK.
Please, ask for any additional info if needed.
Thanks.

BUG-reporter, 2017-03-21:
And another one thing I found during aMule starting:
```
$ amule -o              
 2017-03-21 04:46:37: Logging to stdout enabled
 2017-03-21 04:46:37: Initialising aMule 2.3.2 compiled with wxGTK2 v3.0.2 and Boost 1.62
 2017-03-21 04:46:37: Checking if there is an instance already running...
 2017-03-21 04:46:37: No other instances are running.
 2017-03-21 04:46:37: Завантажено файл довіри, 302 відомі клієнти
 2017-03-21 04:46:37: Зовнішні з'єднання вимкнені у файлі налаштувань.
 2017-03-21 04:46:37: Created Server UDP-Socket at port 4665
 2017-03-21 04:46:37: ListenSocket: Гаразд.
 2017-03-21 04:46:37: Created Client UDP-Socket at port 4672
!2017-03-21 04:46:37: Universal Plug and Play: UPnP Error: CUPnPControlPoint::AddPortMapping: WAN Service not detected.
```

persmule, 2017-05-25:
This also breaks i2p's upnp function: 

> 禁用UPnP设备 (Disabling UPnP devices)
> 
>     OpenWRT router urn:schemas-upnp-org:device:InternetGatewayDevice:2
>     UDN: uuid:e97e6209-3d25-439e-b8b4-582ecc64e37f
> 
> UPnP 支持在您的局域网中没有发现任何支持UPnP的设备。(UPnP support finds no UPnP-available device in your LAN)

persmule, 2017-07-20:
This problem seems resolved on lede 17.01.2.

ecsv, 2019-05-27:
This package was removed in 20987707105119f50487711682bb269762ef43c9. So this ticket should most likely be closed

closed 2023-03-11 by BKPepe