openwrt/routing #293: latest babeld.init script is not working in BB version

acsdkquery, 2017-04-07:
w.r.t commit : 5aa0ece, with procd support, babeld.init script is not working in BB version. 
error : "Procd: babeld::instance1 s in a crash loop 6 crashes, 0 seconds since last crash".
The earlier version without procd script is working fine in BB.

zorun, 2017-04-09:
See 159dc575a6cfe965e6eeaa9e9027fc627bf6069b, which mentions that it needs support for empty arguments in procd.  I haven't been able to find the procd commit it mentions though, so I have no idea when it was introduced.

@acsdkquery BB is quite old, I won't spend time to chase a bug for it.  If you manage to find what exactly is broken, I will happily fix the issue though.

ecsv, 2019-05-27:
This should most likely be closed. @kerneis  @zorun

closed 2020-01-11 by zorun