openwrt/routing #299: quagga-ospf6d: doesn't get routes from other routers

suedwestlicht, 2017-06-04:
Maintainer: acinonyx

Environment:
LEDE Reboot 17.01.1 r3316-7eb58cf109 / LuCI lede-17.01 branch (git-17.100.70571-29fabe2)
TP-Link TL-WR1043N/ND v4

Description:
ospfd of quagga works ok.

ospf6d of quagga recognizes its neighbours but doesn't get routes.
Package: quagga-ospf6d
Version: 1.1.0-1

It seems we have a bad quagga version.
Please see https://lists.gt.net/quagga/users/32153

Would it be possible to package an older version of quagga?

lucize, 2017-08-13:
can you try this #319  ?

suedwestlicht, 2017-08-13:
How can I install the new version?

lucize, 2017-08-13:
`git pull origin pull/319/head` in git root of routing master tree
and after that make a symlink of frr into package/feeds/routing/
`make menuconfig` and you should find it in the routing submenu

lucize, 2017-08-13:
you need to setup a SDK  [hint](https://lede-project.org/docs/guide-developer/the-source-code)
clone the git, resolve dependencies based on your distro then

`./scripts/feeds update -a`
`./scripts/feeds install -a`
clone the PR, make the symlink
`make menuconfig `
select frr packages

lucize, 2017-08-13:
https://lede-project.org/docs/guide-developer/start

suedwestlicht, 2017-08-13:
git clone https://git.lede-project.org/source.git
cd source
./scripts/feeds update -a
./scripts/feeds install -a

*** all went ok ***

Now I am stuck. In which dir shall I clone what?

lucize, 2017-08-13:
cd feeds/routing
git pull origin pull/319/head
ln -s ../../../feeds/routing/frr/ ../../package/feeds/routing/
cd ../..
make menuconfig

suedwestlicht, 2017-08-13:
ok. symlink looks good (resolves)
make menuconfig (shows the correct architecture Atheros AR7xxx/AR9xxx)

now I am too stupid to finde the frr packages. How can I activate them? I guess I have to build quagga's ospf6d only?

do I have to check
[ ] Build the LEDE SDK
[ ] Package the LEDE-based Toolchain
?

lucize, 2017-08-13:
you're not stupid :) !
 on subtarget find your router model and version, confirm on the router box that you have the V4 and select the V4 from subtarget
after that go to `Network` -> `Routing and Redirection` -> `frr` (select it with space to be like * not M and select frr packages that you need

lucize, 2017-08-13:
don't forget to select `LuCI` otherwise you'll not have web interface in the image

suedwestlicht, 2017-08-13:
Oh, I thought I could build the packages for quagga only, not a whole Image.

lucize, 2017-08-13:
`make package/frr/compile V=s -j4`
but then you must have the git version of your installed image
`git checkout <shanumber>` of your installed releas

lucize, 2017-08-13:
it needs libjson-c as a dep, you will find them in bin folder and if you're lucky if the libs haven't change much from your version maybe it will run without checking out the older release

suedwestlicht, 2017-08-13:
It does. I had to build an image first to get some lib stuff. I could build the frr packages.

I use vytsh, zebra, ospfd and ospf6d

frr's zebra, ospfd and ospf6d seem to work fine. Thanks a lot.

vtysh isnt usable out of the box because it doesn't find "more"

```
root@ap-gz:~# vtysh
Hello, this is FRRouting (version 3.0-rc1).
Copyright 1996-2005 Kunihiro Ishiguro, et al.

sh: more: not found
ap-gz# show ipv6 ospf6 route 
sh: more: not found
```

So I have to set
`export VTYSH_PAGER=less`

Then all is fine.

lucize, 2017-08-13:
yes, I saw the more error, but it's still in rc and never been tested on LEDE, hopefully will be fixed
glad it worked out

suedwestlicht, 2017-08-13:
What's next? Will frr replace quagga or will quagga get updated in openwrt or lede?

lucize, 2017-08-13:
the master frr version has more daemons implementations like EIGRP, EVPN so frr seems to be the future, that's why it forked, the packages could coexist, but quagga development seems to be slow
the master version has some issues with cross compile see https://github.com/FRRouting/frr/issues/958

lucize, 2017-08-13:
i'll fix the `more` issue as I found where is it calling

ecsv, 2019-05-27:
@acinonyx