openwrt/routing #311: batmand-adv compilation against cfg80211 driver

hemanth4ap, 2017-07-04:
Hi,
      I am using qsdk (openwrt 12.09) with 3.3.8 kernel version and qualcomm proprietary driver. I want to compile latest batmand-adv package but i am not able to compile the package. so how to compile and any changes is required in makefile or patches please help me.

84ace, 2017-07-06:
What problems do you encounter?

hemanth4ap, 2017-07-10:
we are not able to compile because of dependencies cfg80211. we don't want to use cfg80211 driver so we are removed dependency +kmod-cfg80211 in makefile.

ecsv, 2018-06-13:
cfg80211 is not a hw driver. It is a layer in the kernel to interact with wireless components. Every driver which doesn't support it should be regarded as suspicious (try to avoid using such drivers). And batman-adv requires kmod-cfg80211 package for:

* B.A.T.M.A.N. V's expected throughput query
* the kernel compat(.ko) code provided by the kmod-cfg80211

Even when you don't need B.A.T.M.A.N. V, you still need the compat code. So you cannot just drop the dependency to this package.

----

Replacing the compat code is possible but needs to be done manually. I have an example change at https://github.com/ecsv/routing/commit/2a7ec06b9726a9c3eb0787b241d4b49499adc63c which you can check. But you must understand that this will not be upstreamed because of the previously mentioned requirements. I will also provide no support for this change and may just drop it from my fork at any point.

ecsv, 2019-05-24:
Request is not acceptable for upstream openwrt-routing (as described above). Issue should be closed

closed 2019-05-24 by simonwunderlich