openwrt/routing #334: batman-adv 2017-4: fails to compile with LEDE17.01

SvenRoederer, 2017-12-14:
For freifunk-berlin I recently build the master-feed against LEDE-17.01 (2b66449), which fails with
```
  CC [M]  /mnt/hosts/build/src/openwrt/freifunk/lede/lede/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7620/batman-adv-2017.4/net/batman-adv/../../compat-sources/net/core/skbuff.o
In file included from <command-line>:0:0:
/mnt/hosts/build/src/openwrt/freifunk/lede/lede/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7620/batman-adv-2017.4/compat-hacks.h:324:22: error: expected identifier or '(' before 'struct'
 static inline void  (struct timer_list *timer,
                      ^
```

Batman-adv 2017-4 was merged in https://github.com/openwrt-routing/packages/pull/332

p4u, 2017-12-18:
Same problem here.

nzamps, 2017-12-19:
Try changing the define in combat-hacks.h to :

```
static inline void      timer_setup(struct timer_list *timer,
                               void (*callback)(struct timer_list *),
                               unsigned int flags)
```

SvenRoederer, 2017-12-22:
@nzamps this patch let it build with LEDE-17.01 / https://github.com/lede-project/source/commit/ceea0ac25d987de1f84376405d38d43337d31cf5

ecsv, 2018-01-09:
Can you please retest with the current master.

Because there was some confusion regarding the branches: The lede-17.01 branch never had this problem - the incorrectly committed compat-hacks.h was in the master branch (the one for OpenWrt master)

SvenRoederer, 2018-01-10:
https://github.com/openwrt-routing/packages/commit/467f0b61e219c8e9d9964df604ca7fa87fc7548e already fixed the issue building openwrt-master with lede-17.01.
Our buildbots also created images with openwrt-packages master (f48f332) based on lede-17.01 (https://github.com/openwrt/openwrt/commit/dca4dfacf22e93017c90c308dbeec27a77b4f338)

closed 2018-01-10 by SvenRoederer