openwrt/routing #351: quagga-libzebra issue - vtysh -b flag broken

scaci, 2018-02-28:
- Problem occurs to all device on lede-17.01
- Software versions lede-17.01, package quagga-libzebra - 1.1.0-1
- Steps to reproduce:

      1)Start zebra daemon and other routing daemon, for example ospfd
      2)Configure routing with ospf (or other routing protocol) using vtysh
      3)Activate "service integrated-vtysh-config" in config terminal and save the config
      4)Exit vtysh
      5)Restart quagga daemons
      6)Execute vtysh -b to load Quagga.conf
      7)Quagga does not recognize routing command

[130-vtysh-integrated-configuration.zip](https://github.com/openwrt-routing/packages/files/1767210/130-vtysh-integrated-configuration.zip)

The issue is well known and solved in quagga 1.2.3, bug reference here: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=876049

For quagga 1.1.0 I solved problem using the attached patch

ecsv, 2019-05-27:
@acinonyx

BKPepe, 2023-03-11:
Since this is addressed in the latest OpenWrt versions, I'm closing this. OpenWrt 19.07 is End-of-life this these and that's why I am closing this.

closed 2023-03-11 by BKPepe