openwrt/routing #370: batman-adv: "uci show batman-adv" list no options

SvenRoederer, 2018-04-29:
I'm on lede-17.01 and was trying to run ```uci delete batman-adv.bat0.vis_mode``` to remove the optioin as per https://github.com/openwrt-routing/packages/commit/8a66ed185709df5b9120e602fda760969bebd632.
The command failed with :
> root@Ahof-frieden03:~# uci delete batman-adv.bat0.vis_mode
uci: Entry not found

also ```uci show batman-adv``` only outputs:
> root@Ahof-frieden03:~# uci show batman-adv
batman-adv.bat0=mesh

My config-file is like the default before the mentioned commit. 
I found the error is caused by uci, as the options don't have a value. Would it make sense to add some default values for the options?

mwarning, 2018-05-17:
I think the reason for this is to not place default values there in addition to the defaults used by batman-adv.
Maybe @ecsv can tell why for sure.

ecsv, 2018-05-17:
No, I don't know the reason. @lindnermarek ?

ecsv, 2018-06-13:
Would this change be helpful for you? https://github.com/ecsv/routing/commit/e096c3612d5b25e0112f896e4c56a42b74350747

SvenRoederer, 2018-07-03:
the mentioned change should fix the reported issue.

Even I think @mwarning argument is valid (double bookkeeping), but uci-wise I link the change.

mwarning, 2018-07-03:
@ecsv the change would be helpful. People would not need to lookup the defaults and type of the values (string, bool, number..).

lindnermarek, 2018-07-04:
@mwarning @ecsv The reason for not having any default values is that batman-adv already has all the same default values. What is the advantage of duplicating those values into uci ? Who will keep track of the uci file if those defaults were to change ?

Back then the main idea was behind the 'empty' config file was to steer average joe towards only changing the required settings and not overwhelm with numerous options that don't need to be configured.

If the goal is to document the config file, how about adding examples for all settings which are commented out ?

mwarning, 2018-07-04:
@lindnermarek commenting out all settings is fine with me. But this is @SvenRoederer issue. :-)

SvenRoederer, 2018-07-06:
initially I came to the issue by trying to remove the orphaned "vis_mode" (https://github.com/openwrt-routing/packages/commit/8a66ed185709df5b9120e602fda760969bebd632) via uci-commands to update some existing configurations.

ecsv, 2019-05-24:
There is no /etc/config/batman-adv anymore. And b30eb65d2fc9d99f20e5a893f39c8be055b457e4 commented out all entries + added example values. Issue should be closed

closed 2019-05-24 by simonwunderlich