openwrt/routing #398: Add @wfleurant to repo collaborators

lgierth, 2018-07-25:
Hey friends,

I've not been the most active maintainer of the cjdns packages in a long while. @wfleurant, who was the original author of that bunch of code, is generally a human being extraordinaire and still actively working with cjdns and has offered to take over maintainership of `cjdns` and `luci-app-cjdns`.

Just wanted to get a quick sanity check here before I add him as a collaborator to the repo.

Cool? Thanks!

mwarning, 2018-07-31:
@lgierth as for me, I think it is great to have more people on board. +1

lgierth, 2018-07-31:
Thanks for the positive feedback -- adding @wfleurant now!

(I should have timeboxed this, but 1 week seems plenty for a *quick sanity check*.)

lgierth, 2018-07-31:
@wfleurant yoiu have an invitation, make yourself at home. Best first thing to do is update the maintainer line in the packages :)

mwarning, 2018-08-01:
Don't forget to always make a pull request and try to let someone else to review/merge it (there might be exceptions..). :-)

closed 2018-08-01 by mwarning