openwrt/routing #408: batman-adv Compile Fails with _ADV_DEBUG_LOG=y

jeffsf, 2018-09-06:
Compile fails on 2018-09-05 when selecting advanced debug logging in `make menuconfig`

```
/home/jeff/devel/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/batman-adv-2018.2/net/batman-adv/log.c:220:17: error: 'struct batadv_priv' has no member named 'debug_dir'; did you mean 'debug_log'?
  if (!bat_priv->debug_dir)
                 ^~~~~~~~~
                 debug_log
/home/jeff/devel/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/batman-adv-2018.2/net/batman-adv/log.c:230:49: error: 'struct batadv_priv' has no member named 'debug_dir'; did you mean 'debug_log'?
  d = debugfs_create_file("log", 0400, bat_priv->debug_dir, bat_priv,
                                                 ^~~~~~~~~
                                                 debug_log
```
To replicate:

Debian 9.5 (stretch) build host, which successfully builds ath79 and ar71xx for Archer C7 v2 with a wide range of packages.

```
~/devel/openwrt/feeds/routing$ git log -1 --pretty=one
c394f54b6339fd1c522533a7003000e944bfeff7 (HEAD -> master, origin/master, origin/HEAD) Merge pull request #406 from bluewavenet/nodogsplash-3.2.0
```

master branch, configure ath79, Archer C7 v2, add batman-adv, enable "enable verbose debug logging"

```
make V=s clean download world
```
Note error messages on build failure
```
make[3]: Entering directory '/home/jeff/devel/openwrt/feeds/routing/batman-adv'
rm -f /home/jeff/devel/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/batman-adv-2018.2/.built
touch /home/jeff/devel/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/batman-adv-2018.2/.built_check
cat /dev/null > /home/jeff/devel/openwrt/staging_dir/target-mips_24kc_musl/pkginfo/batman-adv.symvers; for subdir in .; do cat /home/jeff/devel/openwrt/staging_dir/target-mips_24kc_musl/pkginfo/*.symvers 2>/dev/null > /home/jeff/devel/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/batman-adv-2018.2/$subdir/Module.symvers; done
env "batman-adv-y=../../compat-sources/net/core/skbuff.o ../../compat-sources/net/ipv4/igmp.o ../../compat-sources/net/ipv6/mcast_snoop.o " make -j1 -C "/home/jeff/devel/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-4.14.67" ARCH="mips" CROSS_COMPILE="mips-openwrt-linux-musl-" SUBDIRS="/home/jeff/devel/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/batman-adv-2018.2/net/batman-adv" CONFIG_BATMAN_ADV=m CONFIG_BATMAN_ADV_DEBUG=y CONFIG_BATMAN_ADV_DEBUGFS=n CONFIG_BATMAN_ADV_BLA=y CONFIG_BATMAN_ADV_DAT=y CONFIG_BATMAN_ADV_MCAST=y CONFIG_BATMAN_ADV_NC=n CONFIG_BATMAN_ADV_BATMAN_V=y  EXTRA_CFLAGS=" -DCONFIG_BATMAN_ADV=1  -DCONFIG_BATMAN_ADV_DEBUG=1  -DCONFIG_BATMAN_ADV_BLA=1  -DCONFIG_BATMAN_ADV_DAT=1  -DCONFIG_BATMAN_ADV_MCAST=1  -DCONFIG_BATMAN_ADV_BATMAN_V=1 " NOSTDINC_FLAGS="-I/home/jeff/devel/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/batman-adv-2018.2/net/batman-adv -I/home/jeff/devel/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211-backport -I/home/jeff/devel/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211-backport/uapi -I/home/jeff/devel/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211 -I/home/jeff/devel/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211/uapi -I/home/jeff/devel/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/batman-adv-2018.2/include/ -include backport/backport.h -include /home/jeff/devel/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/batman-adv-2018.2/compat-hacks.h" modules
make[4]: Entering directory '/home/jeff/devel/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-4.14.67'
  CC [M]  /home/jeff/devel/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/batman-adv-2018.2/net/batman-adv/log.o
/home/jeff/devel/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/batman-adv-2018.2/net/batman-adv/log.c: In function 'batadv_debug_log_setup':
/home/jeff/devel/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/batman-adv-2018.2/net/batman-adv/log.c:220:17: error: 'struct batadv_priv' has no member named 'debug_dir'; did you mean 'debug_log'?
  if (!bat_priv->debug_dir)
                 ^~~~~~~~~
                 debug_log
/home/jeff/devel/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/batman-adv-2018.2/net/batman-adv/log.c:230:49: error: 'struct batadv_priv' has no member named 'debug_dir'; did you mean 'debug_log'?
  d = debugfs_create_file("log", 0400, bat_priv->debug_dir, bat_priv,
                                                 ^~~~~~~~~
                                                 debug_log
```

jeffsf, 2018-09-06:
Examining `openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/batman-adv-2018.2/net/batman-adv/types.h#218` et seq suggests that `debug_dir` is only a member of the struct when `CONFIG_BATMAN_ADV_DEBUGFS` is set.

Confirmed that setting "enable debugfs support" (which sets `CONFIG_KMOD_BATMAN_ADV_DEBUGFS=y`) allows the compilation to successfully complete.

Not examined if this is just a work-around, or is the root of the issue.

ecsv, 2018-10-22:
The dependency is in the Config.in file: https://github.com/openwrt-routing/packages/blob/9bc898b8f4d715523917f361abfa52ec5570f256/batman-adv/Config.in#L92

This dependency will be dropped in the future to allow batadv_dbg tracing support without debugfs dependency

ecsv, 2018-10-22:
Btw. BATMAN_ADV_DEBUG_LOG is now called BATMAN_ADV_DEBUG. You should update your feed and config.

ecsv, 2019-05-24:
BATMAN_ADV_DEBUG_LOG doesn't exist anymore since 0db66160aaabcd48ebc57c2b8f0ea9157123196d. And both BATMAN_ADV_DEBUG and BATMAN_ADV_DEBUGFS are independent since e6fc697d418be2053fe4b99518f2c704fde6165e.

Issue should be closed.

closed 2019-05-24 by simonwunderlich