openwrt/routing #423: kmod-batman-adv: missing dependency to kmod-crc16

Leo-PL, 2018-10-22:
Since commit 0db66160aaabcd48ebc57c2b8f0ea9157123196d build fails with the following error:

```
make[3]: Entering directory '/mnt/work/priv/openwrt/ar71xx/feeds/routing/batman-adv'
mkdir -p /mnt/work/priv/openwrt/ar71xx/staging_dir/target-mips_24kc_musl/root-ar71xx/stamp
SHELL= flock /mnt/work/priv/openwrt/ar71xx/tmp/.root-copy.flock -c 'cp -fpR /mnt/work/priv/openwrt/ar71xx/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/batman-adv-2018.3/.pkgdir/kmod-batman-adv/. /mnt/work/priv/openwrt/ar71xx/staging_dir/target-mips_24kc_musl/root-ar71xx/'
touch /mnt/work/priv/openwrt/ar71xx/staging_dir/target-mips_24kc_musl/root-ar71xx/stamp/.kmod-
batman-adv_installed
mkdir -p /mnt/work/priv/openwrt/ar71xx/bin/targets/ar71xx/generic/packages /mnt/work/priv/openwrt/ar71xx/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/batman-adv-2018.3/ipkg-mips_24kc/kmod-batman-adv/CONTROL /mnt/work/priv/openwrt/ar71xx/staging_dir/target-mips_24kc_musl/pkginfo
mkdir -p /mnt/work/priv/openwrt/ar71xx/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/batman-adv-2018.3/ipkg-mips_24kc/kmod-batman-adv/etc/modules.d; ( echo "batman-adv"; ) > /mnt/work/priv/openwrt/ar71xx/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/batman-adv-2018.3/ipkg-mips_24kc/kmod-batman-adv/etc/modules.d/batman-adv; 
cp -fpR ./files/. /mnt/work/priv/openwrt/ar71xx/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/batman-adv-2018.3/ipkg-mips_24kc/kmod-batman-adv/
find /mnt/work/priv/openwrt/ar71xx/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/batman-adv-2018.3/ipkg-mips_24kc/kmod-batman-adv -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package kmod-batman-adv is missing dependencies for the following libraries:
crc16.ko
Makefile:99: recipe for target '/mnt/work/priv/openwrt/ar71xx/bin/targets/ar71xx/generic/packages/kmod-batman-adv_4.14.77+2018.3-1_mips_24kc.ipk' failed
make[3]: *** [/mnt/work/priv/openwrt/ar71xx/bin/targets/ar71xx/generic/packages/kmod-batman-adv_4.14.77+2018.3-1_mips_24kc.ipk] Error 1
make[3]: Leaving directory '/mnt/work/priv/openwrt/ar71xx/feeds/routing/batman-adv'
time: package/feeds/routing/batman-adv/compile#0.14#0.03#0.18
package/Makefile:107: recipe for target 'package/feeds/routing/batman-adv/compile' failed
make[2]: *** [package/feeds/routing/batman-adv/compile] Error 2
```

Re-enabling dependency to kmod-lib-crc16 fixes the issue.

Leo-PL, 2018-10-22:
@ecsv

ecsv, 2018-10-22:
See https://github.com/openwrt-routing/packages/commit/9bc898b8f4d715523917f361abfa52ec5570f256

Leo-PL, 2018-10-23:
Fixed, thanks!

closed 2018-10-23 by Leo-PL