openwrt/routing #428: [batadv] continuous log messages from netifd

yogo1212, 2018-11-15:
every now and then, syslog is filled with these messages on boot:

```
Thu Nov 15 11:48:33 2018 daemon.notice netifd: Interface 'mesh' is setting up now
Thu Nov 15 11:48:34 2018 daemon.notice netifd: mesh (17773): ./batadv.sh: eval: line 22: can't create /sys/class/net//batman_adv/mesh_iface: nonexistent directory
Thu Nov 15 11:48:34 2018 daemon.notice netifd: mesh (17773): Command failed: Invalid argument
Thu Nov 15 11:48:34 2018 daemon.notice netifd: mesh (17780): ./batadv.sh: eval: line 31: can't create /sys/class/net//batman_adv/mesh_iface: nonexistent directory
Thu Nov 15 11:48:34 2018 daemon.notice netifd: Interface 'mesh' is now down
Thu Nov 15 11:48:34 2018 daemon.notice netifd: Interface 'mesh' is setting up now
...
```

the messages stop after running `/etc/init.d/network restart`.
`reload` isn't enough. `wifi` also has no effect.

looking at https://github.com/openwrt-routing/packages/blob/master/batman-adv/files/lib/netifd/proto/batadv.sh, it appears the `iface` var is empty.
```
echo "$mesh" > "/sys/class/net/$iface/batman_adv/mesh_iface"
```

yogo1212, 2018-11-16:
`{"proto":"batadv","mesh":"bat0","type":"mesh","mtu":1540}`

ecsv, 2019-05-24:
Looks at first glance like a netifd problem. Since both netifd and the batman-adv configuration changed, can you please check whether it still happens with OpenWrt master?

yogo1212, 2019-05-25:
@ecsv i was thinking the same thing.
the script detecting this error also creates a file indicating that it became active.
~20 test routers have been running the updated version since thursday and none of them have run into this issue.
next week, i'll deploy the new firmware to prod.

looking good so far :+1:

yogo1212, 2019-06-25:
have not seen this again - let's assume the integration fixed it.

closed 2019-06-25 by yogo1212