openwrt/routing #436: [luci-app-bmx6] Incomplete BMX6 graph due to race condition in bmx6json.lua?

rogerpueyo, 2018-12-09:
I have a network with 7 nodes running OpenWrt trunk and BMX6 that create a mesh with the following topology:

[image]

They are connected by cable, they all see and ping/SSH each other. When I ask for the network Graph via luci-app-bmx6's Graph page, I get incomplete topologies, like the following examples:

[image]

[image]

etc., but never the complete topology. Asking for the JSON topology via https://192.168.62.1/cgi-bin/luci/admin/network/BMX6/topology returns the JSON structure, but only the links from one or two (maybe three sometimes) of the nodes appear there. For example:

```JSON
[
  {
    "name": "BMX61",
    "links": [
      {
        "name": "BMX64",
        "rxRate": 100,
        "txRate": 100
      },
      {
        "name": "BMX62",
        "rxRate": 100,
        "txRate": 100
      },
      {
        "name": "BMX63",
        "rxRate": 100,
        "txRate": 100
      }
    ]
  },
  {
    "name": "BMX62",
    "links": [
      {
        "name": "BMX64",
        "rxRate": 100,
        "txRate": 100
      },
      {
        "name": "BMX63",
        "rxRate": 100,
        "txRate": 100
      },
      {
        "name": "BMX61",
        "rxRate": 100,
        "txRate": 100
      }
    ]
  },
  {
    "name": "BMX63",
    "links": [
      {
        "name": "BMX64",
        "rxRate": 100,
        "txRate": 100
      },
      {
        "name": "BMX62",
        "rxRate": 100,
        "txRate": 100
      },
      {
        "name": "BMX61",
        "rxRate": 100,
        "txRate": 100
      }
    ]
  }
]
```

I have observed, however, that changing this block of the bmx6json.lua file https://github.com/openwrt-routing/packages/blob/a7c447903cc77fdf43193d1325b5517e6b97da05/luci-app-bmx6/bmx6/usr/lib/lua/luci/model/bmx6json.lua#L60-62 and adding an "empty" _sys.exec()_ call like:

```lua
	if json_url[1] == "http"  then
		raw,err = wget(url..field,1000)
		sys.exec("")
	else
```

Just by adding this empty sys.exec("") call, I get the whole network topology and the graph is completely represented.

I guess there is some weird race condition in the bmx6json.lua's wget() function which returns the output of the actual "wget" command before it finishes, or something similar, but I have been unable so far to identify where the problem is.

mwarning, 2019-01-11:
Thanks for the analysis.
Closing this because your hot fix was merged: https://github.com/openwrt-routing/packages/pull/437

closed 2019-01-11 by mwarning

rogerpueyo, 2019-01-11:
Well, it's been fixed with a dirty patch, but if it was caused by some race condition, it still is there...

p4u, 2019-01-11:
I'd rater focus on bmx7 stuff :)