openwrt/routing #439: [bird {1|2}] state of package maintenance

n8v8R, 2019-01-12:
@Noltari @tohojo 

from the merge requests in {18.06.|master} it is not very clear to me what the current state of maintaining the bird packages is.

Upstream appears to be at 1.6.5 and 2.0.3 respectively whilst in comparison downstream:

* 1.6.3 in 18.06 for the past year
* 1.6.4. in master for the past 8 months
* no 2.x in 18.06
* 2.0.2 in master for the past 6 months

tohojo, 2019-01-14:
It's as it says in the maintainers field in the Makefiles. Just haven't gotten around to submitting an update for 2.0.3; will do so now, thanks for the reminder! :)

tohojo, 2019-01-14:
https://github.com/openwrt-routing/packages/pull/440

n8v8R, 2019-01-14:
@tohojo Thanks for the feedback/action. Just wondering whether 2.x is planned to be merged into 18.06?

tohojo, 2019-01-14:
The switch to {1,2} packages sadly breaks continuity, so I think pulling that back into 18.06 is likely to break stuff (you can't do an upgrade from opkg if you've already installed the old bird package). So I don't think that's likely to happen, unfortunately...

n8v8R, 2019-01-14:
@tohojo that is also going to happen with the release of the next branch 19.x?

tohojo, 2019-01-14:
n8v8R <notifications@github.com> writes:

> @tohojo that is also going to happen with the release of the next
> branch 19.x?

The 19.x branch should be forked from master, so have the split
packages. And because the split will be part of the base release, we
should be able to bump point releases of Bird after the OpenWrt release
has happened...

n8v8R, 2019-01-14:
@tohojo What I meant was that continuity will break the same between v1.x and v2.x in master, or the next forked branch, seeing that both are present in master. What I am trying to say is that anyone with v1.x in current branch will not be able to automatically upgrade to v2.x in the next branch as opkg will not make the transformation?

Meaning the future scenario looks the same as if 2.x would be merged into current branch.

tohojo, 2019-01-14:
n8v8R <notifications@github.com> writes:

> @tohojo What I meant was that continuity will break the same between
> v1.x and v2.x in master, or the next forked branch, seeing that both
> are present in master. What I am trying to say is that [...]

No, an upgrade to 19.x requires reflashing a new image, either with
packages built in, or manually reinstalling them afterwards. So no
ipkg-managed transition is needed.