openwrt/routing #447: batman-adv behaviour in openwrt 2019.0-0 vs  openwrt-2018.4-0

DanielRIOT, 2019-02-06:
platform : ATH79 / ar71xx ( ar9331 / ar9330 rev 1 )
I'm getting " own address as source address" messages again with the new batman release, with exactly the same configuration on my ar71xx platform ( porting to ATH79 )

* batctl openwrt-2018.4-1 [batman-adv: openwrt-2018.4-0]
`no log messages`

* batctl openwrt-2019.0-0 [batman-adv: openwrt-2019.0-0]			
 `br-bat_clus: received packet on bat0.3 with own address as source address (addr:70:b3:xx:xx:xx:xx, vlan:0)`

some backround : using BATMAN to create an extended lan by bridging a VLAN through it so devices on the particular VLAN can speak through the mesh without knowing about mesh routing

[Ethernet interface] = eth0 ,eth0.3 , 
[br-bat_clus] = bridge between bat0.3 and eth0.3

does anyone have any suggestions for debugging or know what the cause could be  ? ( I've tried turning BLA on and off and it doesn't make a difference )

ecsv, 2019-02-11:
Uhm, did you read the changelog of the last batman-adv release? https://www.open-mesh.org/news/88#batman-adv

We basically had no changes beside netlink stuff and some bugfixes. And the netlink things are only about reporting the current state to userspace. So I would not have any idea what caused it. Could you please bisect the changes and report back what introduced it.

taeasy, 2019-02-14:
CC [M]  /home/main/openwrt-sdk-18.06.2-ramips-mt7620_gcc-7.3.0_musl.Linux-x86_64/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/batman-adv-2019.0/net/batman-adv/bat_iv_ogm.o
/home/main/openwrt-sdk-18.06.2-ramips-mt7620_gcc-7.3.0_musl.Linux-x86_64/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/batman-adv-2019.0/net/batman-adv/bat_iv_ogm.c: In function 'batadv_iv_gw_dump_entry':
/home/main/openwrt-sdk-18.06.2-ramips-mt7620_gcc-7.3.0_musl.Linux-x86_64/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/batman-adv-2019.0/net/batman-adv/bat_iv_ogm.c:2622:2: error: too few arguments to function 'genl_dump_check_consistent'
  genl_dump_check_consistent(cb, hdr);

ecsv, 2019-02-14:
@taeasy: Dont mix openwrt-routing feed for the current openwrt master with the OpenWrt 18.06.x releases. Your OpenWrt 18.06 release just doesn't have the mac80211 package with the compat code for genl_dump_check_consistent

closed 2019-05-20 by DanielRIOT