openwrt/routing #449: batctl: installing multiple variants in parallel

SvenRoederer, 2019-02-18:
In our firmware the batctl-tiny variant gets always installed too, even we explicitly select batctl-default in the packages-list.
```
root@Ahof-frieden03:/usr/sbin# opkg list-installed |grep batctl
batctl-default - 2019.0-0
batctl-tiny - 2019.0-0
```

the package-list is here: https://github.com/freifunk-berlin/firmware/blob/master/packages/default.txt

ecsv, 2019-02-22:
Cannot reproduce this:

    root@OpenWrt:/# opkg list-installed |grep batctl
    batctl-default - 2019.0-1

My (diff)config:

    CONFIG_TARGET_x86=y
    CONFIG_TARGET_x86_64=y
    CONFIG_TARGET_x86_64_Generic=y
    CONFIG_ATH_USER_REGD=y
    CONFIG_BATMAN_ADV_BATMAN_V=y
    CONFIG_BATMAN_ADV_BLA=y
    CONFIG_BATMAN_ADV_DAT=y
    CONFIG_BATMAN_ADV_DEBUG=y
    CONFIG_BATMAN_ADV_DEBUGFS=y
    CONFIG_BATMAN_ADV_MCAST=y
    CONFIG_PACKAGE_ALFRED_VIS=y
    CONFIG_PACKAGE_ATH_DFS=y
    CONFIG_PACKAGE_MAC80211_DEBUGFS=y
    CONFIG_PACKAGE_MAC80211_MESH=y
    CONFIG_PACKAGE_alfred=y
    CONFIG_PACKAGE_batctl-default=y
    CONFIG_PACKAGE_hostapd-common=y
    CONFIG_PACKAGE_iw=y
    CONFIG_PACKAGE_kmod-ath=y
    CONFIG_PACKAGE_kmod-batman-adv=y
    CONFIG_PACKAGE_kmod-cfg80211=y
    CONFIG_PACKAGE_kmod-crypto-crc32c=y
    CONFIG_PACKAGE_kmod-crypto-hash=y
    CONFIG_PACKAGE_kmod-lib-crc16=y
    CONFIG_PACKAGE_kmod-lib-crc32c=y
    CONFIG_PACKAGE_kmod-mac80211=y
    CONFIG_PACKAGE_wireless-regdb=y
    CONFIG_TARGET_IMAGES_PAD=y
    CONFIG_VMDK_IMAGES=y

ecsv, 2019-05-24:
We cannot reproduce the problem and didn't receive any further information how to reproduce it.

simonwunderlich, 2019-05-24:
We cannot reproduce the problem and didn't receive any further information how to reproduce it.

closed 2019-05-24 by simonwunderlich