openwrt/routing #465: Please update package SMCRoute

smcgrath12, 2019-04-02:
Hello, I would like an update for package “SMCROUTE” available via Entware. Basically, update the “ipk” file in Entware for SMCRoute. My router is ASUS RT-AC68. The current ipk file at https://bin.entware.net/armv7sf-k3.2/ for SMCRoute is for version 2.0.0, which is almost 5 years old.

I submitted a request to Entware to update to version 2.4.4 (the latest stable version), but I got a reply of:

“This package came from upstream with no changes.

Please, ask maintainers to update it. The new version will arrive here during next sync.”

I was asked to contact you so that you can get the ball rolling for this update. Any help is highly appreciated. Thank you.

ecsv, 2019-04-05:
@lbschenkel  @lynxis

See https://github.com/troglobit/smcroute/releases/tag/2.4.4 for the newest release

lbschenkel, 2019-04-05:
If a member of the OpenWrt team is reading this, I'm no longer maintaining this package and my name should be removed from it. Hopefully somebody else can take over.

smcgrath12, 2019-04-20:
> @lbschenkel @lynxis
> 
> See https://github.com/troglobit/smcroute/releases/tag/2.4.4 for the newest release

How do I go about using the github version? All I see are TAR files, not IPK. I guess I need to "convert" one of these files to IPK format? Sorry, I don't have a linux background.

lynxis, 2019-04-23:
@lbschenkel ack.

@smcgrath12 do you like to maintain it? If so, I would like to remove this package here and move it over to the "general" packages repository (https://github.com/openwrt/packages).

lynxis, 2019-04-23:
https://github.com/openwrt-routing/packages/pull/467

lynxis, 2019-06-10:
@smcgrath12 there is now a newer version in the packages feeds (2.4.4).

lynxis, 2019-06-10:
Thank you @mwarning for being the new maintainer.

closed 2019-06-10 by lynxis

lbschenkel, 2019-06-11:
Thank you as well for taking over. Unfortunately I'm not using vanilla OpenWrt anymore, and I can't do a good job of maintaining a package if I'm not running it myself.