openwrt/routing #481: babeld: Add stable patches

adrianschmutzler, 2019-06-27:
The last babeld update has been in Nov 2018, and I would like to have the two patches from the 1.8 stable branch included here (https://github.com/jech/babeld/commits/babeld-1.8-branch).

Can I just add them in a PR or is there anything speaking against it (since we have no local babeld patches so far)?
I already do it that way in my own repo ...

kerneis, 2019-06-28:
> Can I just add them in a PR or is there anything speaking against it (since we have no local babeld patches so far)?
>  I already do it that way in my own repo ...

We've had patches in the past I think, we just don't have any that make sense right now but feel free to add a pull request.

Gabriel

adrianschmutzler, 2019-06-28:
https://github.com/openwrt-routing/packages/pull/482

closed 2019-06-28 by adrianschmutzler