openwrt/routing #530: batctl 2019.4: BATMAN mesh bat0 disabled

kledom, 2019-11-13:
I'm having problems with batman-adv after upgrading (soft reset) openwrt to the latest snapshot. Since all my configuration files are the same as before (batctl 2019.3) I suspect a regression bug with the latest update to batctl 2019.4 and kmod-batman-adv - 4.19.82+2019.4.

The problem is that batctl o doesn't return anything and I can't ping other devices in the mesh. I couldnt find any error messages in the logfiles:

```
root@OpenWrt:~# batctl o
root@OpenWrt:~# 
```

```
root@OpenWrt:~# batctl meshif bat0 dc
Error received: No such file or directory
BATMAN mesh bat0 disabled
```

**logread | grep batman**
`Wed Nov 13 13:21:25 2019 kern.info kernel: [   15.400405] batman_adv: B.A.T.M.A.N. advanced openwrt-2019.4-0 (compatibility version 15) loaded`

**logread | grep mesh**
```
Wed Nov 13 13:21:47 2019 kern.info kernel: [   41.042231] br-mesh: port 1(bat0.3333) entered blocking state
Wed Nov 13 13:21:47 2019 kern.info kernel: [   41.048014] br-mesh: port 1(bat0.3333) entered disabled state
Wed Nov 13 13:21:47 2019 kern.info kernel: [   41.288644] br-mesh: port 1(bat0.3333) entered blocking state
Wed Nov 13 13:21:47 2019 kern.info kernel: [   41.294500] br-mesh: port 1(bat0.3333) entered listening state
Wed Nov 13 13:21:47 2019 kern.info kernel: [   41.300621] IPv6: ADDRCONF(NETDEV_UP): br-mesh: link is not ready
Wed Nov 13 13:21:47 2019 daemon.notice netifd: Interface 'mesh' is enabled
Wed Nov 13 13:21:47 2019 daemon.notice netifd: Interface 'mesh' is setting up now
Wed Nov 13 13:21:47 2019 daemon.notice netifd: Interface 'mesh' is now up
Wed Nov 13 13:21:49 2019 kern.info kernel: [   43.310157] br-mesh: port 1(bat0.3333) entered learning state
Wed Nov 13 13:21:51 2019 kern.info kernel: [   45.390158] br-mesh: port 1(bat0.3333) entered forwarding state
Wed Nov 13 13:21:51 2019 kern.info kernel: [   45.396113] br-mesh: topology change detected, propagating
Wed Nov 13 13:21:51 2019 kern.info kernel: [   45.402158] IPv6: ADDRCONF(NETDEV_CHANGE): br-mesh: link becomes ready
Wed Nov 13 13:21:51 2019 daemon.notice netifd: bridge 'br-mesh' link is up
Wed Nov 13 13:21:51 2019 daemon.notice netifd: Interface 'mesh' has link connectivity
Wed Nov 13 13:21:56 2019 user.notice root: ip link set dev mesh0 up
Wed Nov 13 13:21:56 2019 kern.info kernel: [   50.159097] IPv6: ADDRCONF(NETDEV_UP): mesh0: link is not ready
Wed Nov 13 13:21:57 2019 user.notice mac80211: Failed command: iw dev mesh0 set mesh_param mesh_fwding 0
Wed Nov 13 13:21:57 2019 daemon.notice netifd: Interface 'nwi_mesh0' is enabled
```

If you need more information about my settings/configuration please let me know.

Regards
Dominik

**cat /etc/openwrt_release:**
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r11487-dfd8c45a98'
DISTRIB_TARGET='ath79/generic'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r11487-dfd8c45a98'
DISTRIB_TAINTS=''
```

**opkg list-installed**
```
base-files - 207-r11487-dfd8c45a98
batctl-default - 2019.4-0
busybox - 1.31.1-1
cgi-io - 14
dnsmasq - 2.80-14
dropbear - 2019.78-2
firewall - 2019-09-23-daed0cf9-1
fstools - 2019-11-03-f438552e-1
fwtool - 2019-11-09-9d9d4c28-1
getrandom - 2019-06-16-4df34a4d-3
hostapd-common - 2019-08-08-ca8c2bd2-2
ip6tables - 1.8.3-2
iperf3 - 3.7-1
iptables - 1.8.3-2
iw - 5.3-2
iwinfo - 2019-10-16-07315b6f-1
jshn - 2019-10-29-30130391-1
jsonfilter - 2018-02-04-c7e938d6-1
kernel - 4.19.82-1-41e4ac0ff2adb970eba978a9e1409811
kmod-ath - 4.19.82+5.4-rc2-1-2
kmod-ath9k - 4.19.82+5.4-rc2-1-2
kmod-ath9k-common - 4.19.82+5.4-rc2-1-2
kmod-batman-adv - 4.19.82+2019.4-0
kmod-cfg80211 - 4.19.82+5.4-rc2-1-2
kmod-crypto-crc32c - 4.19.82-1
kmod-crypto-hash - 4.19.82-1
kmod-gpio-button-hotplug - 4.19.82-3
kmod-ip6tables - 4.19.82-1
kmod-ipt-conntrack - 4.19.82-1
kmod-ipt-core - 4.19.82-1
kmod-ipt-nat - 4.19.82-1
kmod-ipt-offload - 4.19.82-1
kmod-lib-crc-ccitt - 4.19.82-1
kmod-lib-crc16 - 4.19.82-1
kmod-lib-crc32c - 4.19.82-1
kmod-mac80211 - 4.19.82+5.4-rc2-1-2
kmod-nf-conntrack - 4.19.82-1
kmod-nf-conntrack6 - 4.19.82-1
kmod-nf-flow - 4.19.82-1
kmod-nf-ipt - 4.19.82-1
kmod-nf-ipt6 - 4.19.82-1
kmod-nf-nat - 4.19.82-1
kmod-nf-reject - 4.19.82-1
kmod-nf-reject6 - 4.19.82-1
kmod-nls-base - 4.19.82-1
kmod-phy-ath79-usb - 4.19.82-1
kmod-ppp - 4.19.82-1
kmod-pppoe - 4.19.82-1
kmod-pppox - 4.19.82-1
kmod-slhc - 4.19.82-1
kmod-usb-core - 4.19.82-1
kmod-usb-ehci - 4.19.82-1
kmod-usb-ohci - 4.19.82-1
kmod-usb2 - 4.19.82-1
libblobmsg-json - 2019-10-29-30130391-1
libc - 1.1.24-2
libgcc1 - 8.3.0-2
libip4tc2 - 1.8.3-2
libip6tc2 - 1.8.3-2
libiwinfo-lua - 2019-10-16-07315b6f-1
libiwinfo20181126 - 2019-10-16-07315b6f-1
libjson-c4 - 0.13.1-1
libjson-script - 2019-10-29-30130391-1
liblua5.1.5 - 5.1.5-7
liblucihttp-lua - 2019-07-05-a34a17d5-1
liblucihttp0 - 2019-07-05-a34a17d5-1
libnl-tiny - 2019-10-04-0230d069-1
libopenssl-conf - 1.1.1d-2
libopenssl1.1 - 1.1.1d-2
libpthread - 1.1.24-2
librt - 1.1.24-2
libubox20170601 - 2019-10-29-30130391-1
libubus-lua - 2018-10-06-221ce7e7-1
libubus20170705 - 2018-10-06-221ce7e7-1
libuci20130104 - 2019-11-08-fc417e80-3
libuclient20160123 - 2019-05-30-3b3e368d-1
libustream-openssl20150806 - 2019-11-05-c9b66682-1
libxtables12 - 1.8.3-2
logd - 2019-06-16-4df34a4d-3
lua - 5.1.5-7
luci - git-19.316.39109-95bfb3f-1
luci-app-firewall - git-19.316.39109-95bfb3f-1
luci-app-opkg - git-19.316.39109-95bfb3f-1
luci-base - git-19.316.39109-95bfb3f-1
luci-lib-ip - git-19.316.39109-95bfb3f-1
luci-lib-jsonc - git-19.316.39109-95bfb3f-1
luci-lib-nixio - git-19.316.39109-95bfb3f-1
luci-mod-admin-full - git-19.316.39109-95bfb3f-1
luci-mod-network - git-19.316.39109-95bfb3f-1
luci-mod-status - git-19.316.39109-95bfb3f-1
luci-mod-system - git-19.316.39109-95bfb3f-1
luci-proto-ipv6 - git-19.316.39109-95bfb3f-1
luci-proto-ppp - git-19.316.39109-95bfb3f-1
luci-ssl-openssl - git-19.316.39109-95bfb3f-1
luci-theme-bootstrap - git-19.316.39109-95bfb3f-1
mtd - 25
netifd - 2019-11-12-e15147c2-1
odhcp6c - 2019-01-11-e199804b-16
odhcpd-ipv6only - 2019-10-13-9a4531a4-3
openssl-util - 1.1.1d-2
openwrt-keyring - 2019-07-25-8080ef34-1
opkg - 2019-06-14-dcbc142e-1
ppp - 2.4.7.git-2019-10-04-2
ppp-mod-pppoe - 2.4.7.git-2019-10-04-2
procd - 2019-11-02-f47622e8-4
rpcd - 2019-11-10-77ad0de0-1
rpcd-mod-file - 2019-11-10-77ad0de0-1
rpcd-mod-iwinfo - 2019-11-10-77ad0de0-1
rpcd-mod-luci - 20191108
rpcd-mod-rrdns - 20170710
rssileds - 0.2-2
swconfig - 12
uboot-envtools - 2019.07-1
ubox - 2019-06-16-4df34a4d-3
ubus - 2018-10-06-221ce7e7-1
ubusd - 2018-10-06-221ce7e7-1
uci - 2019-11-08-fc417e80-3
uclient-fetch - 2019-05-30-3b3e368d-1
uhttpd - 2019-08-17-6b03f960-4
urandom-seed - 1.0-1
urngd - 2019-10-21-40f939d5-1
usign - 2019-09-21-f34a383e-1
wireless-regdb - 2019.06.03
wpad-mesh-openssl - 2019-08-08-ca8c2bd2-2
```

kledom, 2019-11-13:
```
root@ROBOT_WS04-AP-2:~# dmesg
[    0.000000] Linux version 4.19.82 (builder@buildhost) (gcc version 8.3.0 (OpenWrt GCC 8.3.0 r11487-dfd8c45a98)) #0 Tue Nov 12 15:43:14 2019
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
[    0.000000] MIPS: machine is Ubiquiti Bullet M
[    0.000000] SoC: Atheros AR7241 rev 1
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 02000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000001ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000001ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000001ffffff]
[    0.000000] On node 0 totalpages: 8192
[    0.000000]   Normal zone: 72 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 8192 pages, LIFO batch:0
[    0.000000] random: get_random_bytes called from start_kernel+0x98/0x4b0 with crng_init=0
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 8120
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 24800K/32768K available (4507K kernel code, 184K rwdata, 1056K rodata, 1220K init, 213K bss, 7968K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] CPU clock: 390.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 9801335621 ns
[    0.000013] sched_clock: 32 bits at 195MHz, resolution 5ns, wraps every 11012737021ns
[    0.007894] Calibrating delay loop... 259.27 BogoMIPS (lpj=1296384)
[    0.084079] pid_max: default: 32768 minimum: 301
[    0.089078] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.095683] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.109700] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.119551] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.125824] pinctrl core: initialized pinctrl subsystem
[    0.134759] NET: Registered protocol family 16
[    0.149670] PCI host bridge /ahb/apb/pcie-controller@180c0000 ranges:
[    0.156188]  MEM 0x0000000010000000..0x0000000013ffffff
[    0.161390]   IO 0x0000000000000000..0x0000000000000000
[    0.194539] PCI host bridge to bus 0000:00
[    0.198726] pci_bus 0000:00: root bus resource [mem 0x10000000-0x13ffffff]
[    0.205561] pci_bus 0000:00: root bus resource [io  0x0000]
[    0.211119] pci_bus 0000:00: root bus resource [??? 0x00000000 flags 0x0]
[    0.217872] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
[    0.225814] pci 0000:00:00.0: [168c:002a] type 00 class 0x028000
[    0.225895] pci 0000:00:00.0: reg 0x10: [mem 0x10000000-0x1000ffff 64bit]
[    0.226080] pci 0000:00:00.0: supports D1
[    0.226095] pci 0000:00:00.0: PME# supported from D0 D1 D3hot
[    0.227697] pci_bus 0000:00: busn_res: [bus 00-ff] end is updated to 00
[    0.227737] pci 0000:00:00.0: BAR 0: assigned [mem 0x10000000-0x1000ffff 64bit]
[    0.240073] clocksource: Switched to clocksource MIPS
[    0.246922] NET: Registered protocol family 2
[    0.252725] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes)
[    0.260481] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[    0.267426] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
[    0.273796] TCP: Hash tables configured (established 1024 bind 1024)
[    0.280348] UDP hash table entries: 256 (order: 0, 4096 bytes)
[    0.286191] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[    0.292973] NET: Registered protocol family 1
[    0.297374] PCI: CLS 0 bytes, default 32
[    0.301841] Crashlog allocated RAM at address 0x1f00000
[... 105 lines cut ...]
[   44.300484] ath: doing EEPROM country->regdmn map search
[   44.305814] ath: country maps to regdmn code: 0x37
[   44.310638] ath: Country alpha2 being used: DE
[   44.315093] ath: Regpair used: 0x37
[   44.318593] ath: regdomain 0x8114 dynamically updated by user
[   44.350151] br-mesh: port 1(bat0.3333) entered learning state
[   46.430120] br-mesh: port 1(bat0.3333) entered forwarding state
[   46.436074] br-mesh: topology change detected, propagating
[   46.442098] IPv6: ADDRCONF(NETDEV_CHANGE): br-mesh: link becomes ready
[   48.158922] IPv6: ADDRCONF(NETDEV_UP): wlan0-1: link is not ready
[   48.246979] br-lan: port 2(wlan0-1) entered blocking state
[   48.252581] br-lan: port 2(wlan0-1) entered disabled state
[   48.258603] device wlan0-1 entered promiscuous mode
[   48.785113] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0-1: link becomes ready
[   48.792016] br-lan: port 2(wlan0-1) entered blocking state
[... cut ...]
[   48.827107] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   50.279614] IPv6: ADDRCONF(NETDEV_UP): mesh0: link is not ready
[   51.316781] br-lan: port 2(wlan0-1) entered disabled state
[   52.979859] do_page_fault(): sending SIGSEGV to wpa_supplicant for invalid read access from 000005b8
[   52.989113] epc = 004840c9 in wpad[400000+b4000]
[   52.993821] ra  = 004840b3 in wpad[400000+b4000]
[   54.730591] br-lan: port 2(wlan0-1) entered blocking state
[   54.736159] br-lan: port 2(wlan0-1) entered forwarding state
```

kledom, 2019-11-13:
Downgrading to 19.07.0-rc1 fixed the issue.

adrianschmutzler, 2019-11-13:
If nobody sees anything obvious, you could try to find the commit breaking it in master, starting from the 19.07 branch ...

kledom, 2019-11-14:
I probably don t have the right infrastructure to find the causing commit. Let s wait for sombody confirming the problem.
I find the line
`[   52.979859] do_page_fault(): sending SIGSEGV to wpa_supplicant for invalid read access from 000005b8`
a little bit suspicious. Maybe the bug is in wpad-mesh-openssl?

adrianschmutzler, 2019-11-14:
And this line is not present on your "working configuration"?

adrianschmutzler, 2019-11-14:
FYI: https://github.com/openwrt/openwrt/commit/000b7687bc50be5b0f1161f4bf8ceb85c495c395

Sounds like it could be related.

simonwunderlich, 2019-11-14:
If wpa_supplicant (a userspace program) segfaults, then I don't see how batman-adv (a kernel module) could cause that. I'd suggest to file a bug against wpad.

kledom, 2019-11-15:
> 
> 
> And this line is not present on your "working configuration"?

No. This is the output of dmesg on the working configuration.

```
root@OpenWrt:~# dmesg
[    0.000000] Linux version 4.14.151 (builder@buildhost) (gcc version 7.4.0 (OpenWrt GCC 7.4.0 r10649-c4fdb377a2)) #0 Tue Nov 5 23:26:40 2019
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
[    0.000000] MIPS: machine is Ubiquiti Bullet M
[    0.000000] SoC: Atheros AR7241 rev 1
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 02000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000001ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000001ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000001ffffff]
[    0.000000] On node 0 totalpages: 8192
[    0.000000] free_area_init_node: node 0, pgdat 804f9f90, node_mem_map 810065a0
[    0.000000]   Normal zone: 64 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 8192 pages, LIFO batch:0
[    0.000000] random: get_random_bytes called from 0x804fd740 with crng_init=0
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 8128
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] PID hash table entries: 128 (order: -3, 512 bytes)
[    0.000000] Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 25852K/32768K available (4025K kernel code, 159K rwdata, 532K rodata, 1228K init, 211K bss, 6916K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] CPU clock: 390.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 9801335621 ns
[    0.000013] sched_clock: 32 bits at 195MHz, resolution 5ns, wraps every 11012737021ns
[    0.007898] Calibrating delay loop... 259.27 BogoMIPS (lpj=1296384)
[    0.084096] pid_max: default: 32768 minimum: 301
[    0.088998] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.095607] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.108466] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.118296] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.124476] pinctrl core: initialized pinctrl subsystem
[    0.133053] NET: Registered protocol family 16
[    0.141368] PCI host bridge /ahb/apb/pcie-controller@180c0000 ranges:
[    0.147805]  MEM 0x0000000010000000..0x0000000013ffffff
[    0.153061]   IO 0x0000000000000000..0x0000000000000000
[    0.182154] PCI host bridge to bus 0000:00
[    0.186326] pci_bus 0000:00: root bus resource [mem 0x10000000-0x13ffffff]
[    0.193165] pci_bus 0000:00: root bus resource [io  0x0000]
[    0.198720] pci_bus 0000:00: root bus resource [??? 0x00000000 flags 0x0]
[    0.205467] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
[    0.213404] pci 0000:00:00.0: [168c:002a] type 00 class 0x028000
[    0.213485] pci 0000:00:00.0: reg 0x10: [mem 0x10000000-0x1000ffff 64bit]
[    0.213668] pci 0000:00:00.0: supports D1
[    0.213682] pci 0000:00:00.0: PME# supported from D0 D1 D3hot
[    0.214029] pci_bus 0000:00: busn_res: [bus 00-ff] end is updated to 00
[    0.214066] pci 0000:00:00.0: BAR 0: assigned [mem 0x10000000-0x1000ffff 64bit]
[    0.225838] clocksource: Switched to clocksource MIPS
[    0.232359] NET: Registered protocol family 2
[    0.238082] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[... 111 lines cut ...]
[   34.534050] br-mesh: port 1(bat0.3333) entered listening state
[   34.540223] IPv6: ADDRCONF(NETDEV_UP): br-mesh: link is not ready
[   36.585925] br-mesh: port 1(bat0.3333) entered learning state
[   38.665927] br-mesh: port 1(bat0.3333) entered forwarding state
[   38.671879] br-mesh: topology change detected, propagating
[   38.678247] IPv6: ADDRCONF(NETDEV_CHANGE): br-mesh: link becomes ready
[   39.200210] IPv6: ADDRCONF(NETDEV_UP): wlan0-1: link is not ready
[   39.215659] br-lan: port 2(wlan0-1) entered blocking state
[   39.221278] br-lan: port 2(wlan0-1) entered disabled state
[   39.227264] device wlan0-1 entered promiscuous mode
[   39.232309] br-lan: port 2(wlan0-1) entered blocking state
[   39.237884] br-lan: port 2(wlan0-1) entered forwarding state
[   39.244477] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   39.687763] br-lan: port 2(wlan0-1) entered disabled state
[   39.713535] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0-1: link becomes ready
[   39.720425] br-lan: port 2(wlan0-1) entered blocking state
[   39.725997] br-lan: port 2(wlan0-1) entered forwarding state
[   39.976620] IPv6: ADDRCONF(NETDEV_UP): mesh0: link is not ready
[   40.746256] br-lan: port 2(wlan0-1) entered disabled state
[   42.352241] IPv6: ADDRCONF(NETDEV_CHANGE): mesh0: link becomes ready
[   42.601399] batman_adv: bat0: Adding interface: mesh0
[   42.606758] batman_adv: bat0: Interface activated: mesh0
[   42.674750] br-lan: port 2(wlan0-1) entered blocking state
[   42.680348] br-lan: port 2(wlan0-1) entered forwarding state
```

ecsv, 2019-11-15:
This doesn't seem to be the fault of batman-adv. If the mesh0 is not enabled correctly and thus not added to bat0 then bat0 cannot do anything. I am assuming here that you use wpa_supplicant to create an encrypted meshpoint or IBSS interface. And that wpa_supplicant crashed and thus the mesh0 interface wasn't initialized correctly.

The first important message missing in the broken setup is ``batman_adv: bat0: Adding interface: mesh0``. Which should have been triggered by netifd (scripts) when mesh0 would have been setup correctly.

@simonwunderlich maybe you want to close this ticket at some point

kledom, 2019-11-15:
Ok. Still thanks for your support :)

closed 2019-11-15 by kledom