openwrt/routing #545: miniupnpd: Unable to add IPv6 Firewall rules (AddPinhole) in version 2.0

ffrediani, 2020-01-24:
When trying to add IPv6 Firewall rules on OpenWrt 17.01.6 which runs miniupnpd version 2.0.20170421-2 it doesn't seem to work as at all.

I used the miniupnpc client from http://miniupnp.free.fr/ and run the following example command:
```
upnpc -6 -A "" "" 2001:db8:1234::5678 12345 tcp 300
```
The two error messages I get are:
```
AddPinhole([]: -> [2001:db8:1234::5678]:12345) failed with code -3 (UnknownError)
AddPinhole([]: -> [2001:db8:1234::5678]:12345) failed with code 401 (Invalid Action)
```
When looking at the miniupnpd daemon logs in the router I see:
"daemon.notice miniupnpd[3622]: SoapMethod: Unknown: AddPinhole"

Although for miniupnpd version 2.0 doesn't have the option specifically to choose IGD version to advertise - v1 or v2 - (differently from miniupnpd 2.1 in OpenWrt 18.06 which does), it is possible to see in the logs that miniupnpd 2.0 listen also on a IPv6 Address.
As support for IPv6 was added to miniupnpd back in 2012 I suspect that maybe binary has not been compiled with --igd2 option as discussed in their forum.

As this package is optional and installed only after the generic factory/sysupgrade firmware deployment would it be the case to upgrade only the version of miniupnpd package to 2.1 used in 17.01 repository so when when it gets installed via opkg it has the same behaviour of 18.06 which works better for adding IPv6 rules dynamically ?

After all 17.01 has full IPv6 support and as such UPnP/PCP should work properly for more coming applications that require that for incoming IPv6 connections and to avoid lay people to add rules manually. Even though there are issues to maintain these rules when IPv6 Prefix Delegation received from the ISP are dynamic.

Maintainer: nobody
Compile tested: ar71xx
Run tested: TP-Link TL-WDR4300 v1

Neustradamus, 2022-08-09:
@ffrediani: What is the situation today?

BKPepe, 2023-03-11:
This package was removed in this commit https://github.com/openwrt/routing/commit/20987707105119f50487711682bb269762ef43c9 and moved to the [packages repository](https://github.com/openwrt/packages). Closing this.

closed 2023-03-11 by BKPepe

ffrediani, 2023-03-11:
As a noted this doesn't seem to have been fixed for that specific version, in the case those using it seek for information.

BKPepe, 2023-03-11:
I am so sorry, but OpenWrt 19.07 is end of life for some time and you should use the supported versions.

ffrediani, 2023-03-11:
When this issue was opened it was still the case that was not end of life and ideally should have been fixed before it. Also there is a significant jump between 19.xx and 18.xx and onwards and these newer versions don't always work well or stable in those hardware where 19.xx used to work, so it is not just as easy to use a newer version on it just because it exists.

The propose of issue was to fix something important in order to make IPv6 work 100% even on these devices that would remain using 19.xx forever because of these limitation.

BKPepe, 2023-03-11:
As always, any pull requests are appreciated and if it was not fixed before, it is not going to be fixed today. OpenWrt 19.07 already has many security vulnerabilities caused by the packages which were shipped there and they are not going to be updated. So if you have only this issue that miniupnpd is not working well for you, I think you have bigger problems than this.

ffrediani, 2023-03-11:
Don't take it to sensitive asking for a pull request from the same person who created the issue. It was enough work to troubleshoot, understand and gather every detail together in order to put it in that way.
And that was not working well "for me" only, but for anyone who is still using 19.xx on older devices and will continue using for a while.