openwrt/routing #547: gpsd changes result in olsrd being unbuildable

RussellSenior, 2020-02-11:
gpsd interface changes in the way timestamping works has broken the position update distribution plugin in olsrd. The workaround (I don't need pud) is to remove pud, but it really ought to be patched up.  Here's my horrible hack:

````
index 5f4e1e3..ce82027 100644
--- a/olsrd/Makefile
+++ b/olsrd/Makefile
@@ -116,12 +116,6 @@ define Package/olsrd-mod-pgraph
   TITLE:=output network topology for pgraph
 endef
 
-define Package/olsrd-mod-pud
-  $(call Package/olsrd/template)
-  DEPENDS:=olsrd +libgps
-  TITLE:=Position Update Distribution plugin
-endef
-
 define Package/olsrd-mod-quagga
   $(call Package/olsrd/template)
   DEPENDS:=olsrd
@@ -152,10 +146,6 @@ define Package/olsrd-mod-watchdog
   TITLE:=Watchdog plugin
 endef
 
-define Package/olsrd-mod-pud/conffiles
-/etc/olsrd.d/olsrd.pud.position.conf
-endef
-
 define Package/olsrd-mod-secure/conffiles
 /etc/olsrd.d/olsrd_secure_key
 endef
@@ -170,7 +160,7 @@ MAKE_FLAGS+= \
        DESTDIR="$(PKG_INSTALL_DIR)" \
        STRIP="true" \
        INSTALL_LIB="true" \
-       SUBDIRS="arprefresh bmf dot_draw dyn_gw dyn_gw_plain httpinfo jsoninfo netjson mdns nameservice p2pd pgraph pud quagga secure sgwdynspeed txtinfo watchdog"
+       SUBDIRS="arprefresh bmf dot_draw dyn_gw dyn_gw_plain httpinfo jsoninfo netjson mdns nameservice p2pd pgraph quagga secure sgwdynspeed txtinfo watchdog"
 
 define Build/Compile
        $(call Build/Compile/Default,all)
@@ -252,15 +242,6 @@ define Package/olsrd-mod-pgraph/install
        $(INSTALL_BIN) $(PKG_BUILD_DIR)/lib/pgraph/olsrd_pgraph.so.* $(1)/usr/lib/
 endef
 

````

cotequeiroz, 2020-02-14:
I've made a patch and opened https://github.com/OLSR/olsrd/pull/79, but I don't use pud, so I don't know if it is working as expected or not.  If someone can test it, then I'll open a PR here as well.

SvenRoederer, 2020-06-07:
there is also a discussion in the openwrt Mailinglist going on: http://lists.infradead.org/pipermail/openwrt-devel/2020-June/023841.html

closed 2020-06-18 by zorun