openwrt/routing #548: luci-app-bmx6 some content not loaded with new luci

pedro-nonfree, 2020-02-21:
testing `luci-app-bmx6` in 19.07.1

[image]

[image]

[image]

in `cgi-bin/luci/admin/network/BMX6/Configuration`

```
Failed to execute cbi dispatcher target for entry '/admin/network/BMX6/Configuration'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:756: module 'luci.cbi' not found:
	no field package.preload['luci.cbi']
	no file './luci/cbi.lua'
	no file '/usr/share/lua/luci/cbi.lua'
	no file '/usr/share/lua/luci/cbi/init.lua'
	no file '/usr/lib/lua/luci/cbi.lua'
	no file '/usr/lib/lua/luci/cbi/init.lua'
	no file './luci/cbi.so'
	no file '/usr/lib/lua/luci/cbi.so'
	no file '/usr/lib/lua/loadall.so'
	no file './luci.so'
	no file '/usr/lib/lua/luci.so'
	no file '/usr/lib/lua/loadall.so'
stack traceback:
	[C]: in function 'require'
	/usr/lib/lua/luci/dispatcher.lua:756: in function </usr/lib/lua/luci/dispatcher.lua:755>
```

- this last one solved installing `luci-compat` src https://github.com/openwrt/packages/issues/10492#issuecomment-551206255

extra src https://lwn.net/Articles/809159/

```
# opkg list-installed | grep bmx6
bmx6 - r2015080701-4
bmx6-json - r2015080701-4
bmx6-sms - r2015080701-4
bmx6-table - r2015080701-4
bmx6-uci-config - r2015080701-4
luci-app-bmx6 - 5
```

web browser console says for that 3 tabs TypeError: `st is null`

```
root@masked:/usr/lib/lua/luci# grep -ir "TablePooler" *
view/bmx6/nodes_j.htm:		new TablePooler(5,"/cgi-bin/bmx6-info", {'$neighbours':''}, "descriptions_table", function(st){
view/bmx6/status_j.htm:		new TablePooler(5,"/cgi-bin/bmx6-info", {'$info':''}, "status_table", function(st){
view/bmx6/tunnels_j.htm:		new TablePooler(5,"/cgi-bin/bmx6-info", {'$tunnels':''}, "descriptions_table", function(st){
```

jow-, 2020-02-21:
@pedro-nonfree - `st` is usually supposed to contain the parsed JSON of the HTTP response. If it is `null`, then the JSON decode failed on the response. This can happen if the invoked CGI script replied with an error for example...

Can you try to open http://192.168.1.1/cgi-bin/bmx6-info?$neighbours in your browser (replace IP address if needed) and see if you get a JSON response or something else like a server error?

pedro-nonfree, 2020-02-21:
hi @jow- , thanks for answering, with http://10.1.59.65/cgi-bin/bmx6-info?$neighbours I got plenty of data which looks right

jow-, 2020-02-21:
Can you paste that data here?

pedro-nonfree, 2020-02-21:
```
{ "neighbours": [ { "originators": [ { "name": "BCNBiscaia450", "blocked": 0, "primaryIp": "fd66:66:66:9:f29f:c2ff:fe54:4c44", "routes": 1, "viaIp": "fe80::26a4:3cff:fe76:67fc", "viaDev": "wif0_t", "metric": "24512K", "lastDesc": 3286, "lastRef": 3 },{ "name": "BCNCoroleu29", "blocked": 0, "primaryIp": "fd66:66:66:8:26a4:3cff:fe78:62ba", "routes": 1, "viaIp": "fe80::26a4:3cff:fe76:67fc", "viaDev": "wif0_t", "metric": "37859K", "lastDesc": 3287, "lastRef": 3 },{ "name": "BCNDoctorSantPonc91", "blocked": 0, "primaryIp": "fd66:66:66:8:26a4:3cff:fe4a:e1ad", "routes": 1, "viaIp": "fe80::26a4:3cff:fe76:67fc", "viaDev": "wif0_t", "metric": "26418K", "lastDesc": 3288, "lastRef": 0 },{ "name": "BCNErcilla68", "blocked": 0, "primaryIp": "fd66:66:66:8:2c0:caff:fe93:4b36", "routes": 1, "viaIp": "fe80::26a4:3cff:fe76:67fc", "viaDev": "wif0_t", "metric": "28084K", "lastDesc": 3289, "lastRef": 0 },{ "name": "BCNFabraiPuig98", "blocked": 0, "primaryIp": "fd66:66:66:9:feec:daff:fe58:a515", "routes": 1, "viaIp": "fe80::26a4:3cff:fe76:67fc", "viaDev": "wif0_t", "metric": "39816K", "lastDesc": 3286, "lastRef": 3 },{ "name": "BCNLLobetivallosera41", "blocked": 0, "primaryIp": "fd66:66:66:8:de9f:dbff:fe2a:359e", "routes": 1, "viaIp": "fe80::26a4:3cff:fe76:67fc", "viaDev": "wif0_t", "metric": "28112K", "lastDesc": 3288, "lastRef": 3 },{ "name": "BCNLaCinetika", "blocked": 0, "primaryIp": "fd66:66:66:b:26a4:3cff:fe78:5847", "routes": 1, "viaIp": "fe80::26a4:3cff:fe76:67fc", "viaDev": "wif0_t", "metric": "39446K", "lastDesc": 3287, "lastRef": 3 },{ "name": "BCNLaFabraRd2", "blocked": 0, "primaryIp": "fd66:66:66:5:de9f:dbff:fe05:8523", "routes": 1, "viaIp": "fe80::26a4:3cff:fe76:67fc", "viaDev": "wif0_t", "metric": "38975K", "lastDesc": 3290, "lastRef": 1 },{ "name": "BCNLaFabraRd3", "blocked": 0, "primaryIp": "fd66:66:66:6:de9f:dbff:feeb:c9b0", "routes": 1, "viaIp": "fe80::26a4:3cff:fe76:67fc", "viaDev": "wif0_t", "metric": "38488K", "lastDesc": 3290, "lastRef": 0 },{ "name": "BCNLlenguadoc10", "blocked": 0, "primaryIp": "fd66:66:66:8:26a4:3cff:fe76:67fc", "routes": 1, "viaIp": "fe80::26a4:3cff:fe76:67fc", "viaDev": "wif0_t", "metric": "54677K", "lastDesc": 3292, "lastRef": 0 },{ "name": "BCNLlenguadoc17Rack", "blocked": 0, "primaryIp": "fd66:66:66:e:1e7d:96ff:fe40:49f", "routes": 1, "viaIp": "fe80::26a4:3cff:fe76:67fc", "viaDev": "wif0_t", "metric": "38975K", "lastDesc": 3285, "lastRef": 1 },{ "name": "BCNLlenguadoc17rooftop", "blocked": 0, "primaryIp": "fd66:66:66:6:26a4:3cff:fe8d:a5cc", "routes": 1, "viaIp": "fe80::26a4:3cff:fe76:67fc", "viaDev": "wif0_t", "metric": "39816K", "lastDesc": 3288, "lastRef": 4 },{ "name": "BCNLlenguadoc17tester", "blocked": 0, "primaryIp": "fd66:66:66:9:1ae8:29ff:fe87:9fad", "routes": 0, "viaIp": "::", "viaDev": "---", "metric": "128G", "lastDesc": 3340, "lastRef": 0 },{ "name": "BCNLlenguadoc21", "blocked": 0, "primaryIp": "fd66:66:66:b:26a4:3cff:fe78:57ee", "routes": 1, "viaIp": "fe80::26a4:3cff:fe76:67fc", "viaDev": "wif0_t", "metric": "39816K", "lastDesc": 3286, "lastRef": 3 },{ "name": "BCNLlenguadoc25", "blocked": 0, "primaryIp": "fd66:66:66:8:26a4:3cff:fe00:f28", "routes": 1, "viaIp": "fe80::26a4:3cff:fe76:67fc", "viaDev": "wif0_t", "metric": "39446K", "lastDesc": 3287, "lastRef": 3 },{ "name": "BCNLlenguadoc74", "blocked": 0, "primaryIp": "fd66:66:66:9:822a:a8ff:fee4:b8a9", "routes": 1, "viaIp": "fe80::26a4:3cff:fe76:67fc", "viaDev": "wif0_t", "metric": "39332K", "lastDesc": 3289, "lastRef": 5 },{ "name": "BCNMercadal5", "blocked": 0, "primaryIp": "fd66:66:66:9:822a:a8ff:fee4:b85e", "routes": 1, "viaIp": "fe80::26a4:3cff:fe76:67fc", "viaDev": "wif0_t", "metric": "29436K", "lastDesc": 3288, "lastRef": 3 },{ "name": "BCNMeridiana550Rd1", "blocked": 0, "primaryIp": "fd66:66:66:6:227:22ff:feb9:e3ad", "routes": 1, "viaIp": "fe80::26a4:3cff:fe76:67fc", "viaDev": "wif0_t", "metric": "30074K", "lastDesc": 3288, "lastRef": 3 },{ "name": "BCNMeridiana550Rd2", "blocked": 0, "primaryIp": "fd66:66:66:6:215:6dff:fee3:2446", "routes": 1, "viaIp": "fe80::26a4:3cff:fe76:67fc", "viaDev": "wif0_t", "metric": "28598K", "lastDesc": 3289, "lastRef": 1 },{ "name": "BCNMonges40", "blocked": 0, "primaryIp": "fd66:66:66:9:feec:daff:fe9a:9db0", "routes": 1, "viaIp": "fe80::26a4:3cff:fe76:67fc", "viaDev": "wif0_t", "metric": "33476K", "lastDesc": 3290, "lastRef": 0 },{ "name": "BCNNouBarris23Rd2", "blocked": 0, "primaryIp": "fd66:66:66:7:26a4:3cff:feb8:450d", "routes": 1, "viaIp": "fe80::26a4:3cff:fe76:67fc", "viaDev": "wif0_t", "metric": "28782K", "lastDesc": 3287, "lastRef": 3 },{ "name": "BCNPonsiGallarza24", "blocked": 0, "primaryIp": "fd66:66:66:8:de9f:dbff:fe5a:e1d", "routes": 1, "viaIp": "fe80::26a4:3cff:fe76:67fc", "viaDev": "wif0_t", "metric": "30074K", "lastDesc": 3287, "lastRef": 3 },{ "name": "BCNRamonBatlle57", "blocked": 0, "primaryIp": "fd66:66:66:8:26a4:3cff:fe64:b2f", "routes": 1, "viaIp": "fe80::26a4:3cff:fe76:67fc", "viaDev": "wif0_t", "metric": "31301K", "lastDesc": 3286, "lastRef": 3 },{ "name": "BCNRamonBatlle9", "blocked": 0, "primaryIp": "fd66:66:66:8:26a4:3cff:fee8:c6bd", "routes": 1, "viaIp": "fe80::26a4:3cff:fe76:67fc", "viaDev": "wif0_t", "metric": "31641K", "lastDesc": 3286, "lastRef": 2 },{ "name": "BCNSantillana7", "blocked": 0, "primaryIp": "fd66:66:66:a:618:d6ff:fe6e:bf22", "routes": 1, "viaIp": "fe80::26a4:3cff:fe76:67fc", "viaDev": "wif0_t", "metric": "38134K", "lastDesc": 3290, "lastRef": 0 },{ "name": "STGTorreRoja15", "blocked": 0, "primaryIp": "fd66:66:66:b:2c0:caff: [... cut ...] pDivisor": 64, "rxExpNumerator": 64, "pathWindow": 5, "pathLounge": 1, "pathRegression": 1, "hopPenalty": 0 } ] }, { "HNA6_EXTENSION": [ { "prefixlen": 128, "address": "fd66:66:66:b:2c0:caff:fe63:385a" }, { "prefixlen": 128, "address": "fd66:66:66:ff00:2c0:caff:fe63:385a" } ] }, { "TUN6_EXTENSION": [ { "localIp": "fd66:66:66:ff00:2c0:caff:fe63:385a" } ] }, { "TUN4IN6_NET_EXTENSION": [ { "tun6Id": 0, "rtype": 0, "bandwidth": "128G", "networklen": 27, "network": "10.1.60.96" } ] } ] } } ]} ] }
```

pedro-nonfree, 2020-02-21:
I suspect the problem is with the bmx6 javascript part about TablePooler

```
        function TablePooler (time, jsonurl, getparams, table_id, callback) {    
                this.table = document.getElementById(table_id);        
                this.callback = callback;                                                 
                this.jsonurl = jsonurl;                                                                      
                this.getparams = getparams;                                         
                this.time = time;                                                
                                                                          
                this.clear = function(){                                                  
                        /* clear all rows */                                                                 
                        while( this.table.rows.length > 1 ) this.table.deleteRow(1);
                }                                                                
                this.start = function(){                                  
                        XHR.poll(this.time, this.jsonurl, this.getparams, function(x, st){
                                var data = this.callback(st);                                                
                                var content, tr, td;                                
                                this.clear();                                    
                                for (var i = 0; i < data.length; i++){     
                                        tr = this.table.insertRow(-1);                    
                                        tr.className = 'cbi-section-table-row cbi-rowstyle-' + ((i % 2) + 1);
                                                                                
                                        for (var j = 0; j < data[i].length; j++){
                                                td = tr.insertCell(-1);   
                                                if (data[i][j].length == 2) {
                                                        td.colSpan = data[i][j][1];                          
                                                        content = data[i][j][0];                    
                                                }                                
                                                else content = data[i][j];
                                                td.innerHTML = content;      
                                        }                                          
                                }                                               
                        }.bind(this));             
                }                                                         
                                                                       
                                                                             
                this.start();                                                          
        }
```

pedro-nonfree, 2020-02-21:
this way the poll gets data!

```
XHR.poll(500, "localhost/cgi-bin/bmx6-info", {"$neighbours":""}, function(st){
  console.log(st)                                       
})
```

jow-, 2020-02-21:
So the trailing slash is the problem?

pedro-nonfree, 2020-02-21:
pull request done, can we have this for 19.07.2 ?

closed 2020-02-21 by jow-

pedro-nonfree, 2020-02-21:
@jow a question, while this is not in next stable, I would like to patch it in my custom firmware (also as an experiment)

I extracted the patch with `git format-patch -1 HEAD`

and I added it to `package/feeds/routing/patches/0001-luci-app-bmx6-bugfix-format-to-query-bmx6-info.patch`

but the patch is not getting applied, why?

pedro-nonfree, 2020-02-21:
@jow- I missed this one

```
vi /usr/lib/lua/luciview/admin_status/index/neighbours_simple.htm
     XHR.poll(5, '/cgi-bin/bmx6-info', { 'neighbours': '' },
```

PR #550 (2020-02-21): luci-app-bmx6: bugfix format to query bmx6-info
- affected three tab luci templates (even with luci-compat installed)
- fixes #548
- as jow said in #548:
  - it might be that modern LuCI escapes it as %24 while old versions passed it
    on as-is
  - bmx6-info removes it anyway
    https://github.com/openwrt-routing/packages/blob/master/luci-app-bmx6/bmx6/www/cgi-bin/bmx6-info#L22
you can probably just remove it in all three tab templates
commit b92f82ad (2020-02-21): luci-app-bmx6: bugfix format to query bmx6-info

- affected three tab luci templates (even with luci-compat installed)
- fixes #548
- as jow said in #548:
  - it might be that modern LuCI escapes it as %24 while old versions passed it
    on as-is
  - bmx6-info removes it anyway
    https://github.com/openwrt-routing/packages/blob/master/luci-app-bmx6/bmx6/www/cgi-bin/bmx6-info#L22
you can probably just remove it in all three tab templates
commit f1b0476e (2020-02-21): luci-app-bmx6: bugfix format to query bmx6-info

- affected three tab luci templates (even with luci-compat installed)
- fixes #548
- as jow said in #548:
  - it might be that modern LuCI escapes it as %24 while old versions passed it
    on as-is
  - bmx6-info removes it anyway
    https://github.com/openwrt-routing/packages/blob/master/luci-app-bmx6/bmx6/www/cgi-bin/bmx6-info#L22
you can probably just remove it in all three tab templates

(cherry picked from commit b92f82ad1f8bb572395fc1b843d409fafb3fc0d3)