openwrt/routing #549: compile error

hyird, 2020-02-21:
make[6]: Leaving directory '/home/hyird/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/olsrd-0.9.7/lib/pud/wireformat'
[CC] src/gpsdclient.c
src/gpsdclient.c: In function 'nmeaInfoFromGpsd':
src/gpsdclient.c:291:101: warning: format '%f' expects argument of type 'double', but argument 12 has type 'timespec_t {aka struct timespec}' [-Wformat=]
syslog(LOG_INFO, "Using %s device%s%s on %s in %s mode at %u baud (%u%c%u), refresh time %.3f (min. %.3f)", //
~~~^
src/gpsdclient.c:301:13:
dev->cycle, //
~~~~~~~~~~
src/gpsdclient.c:291:112: warning: format '%f' expects argument of type 'double', but argument 13 has type 'timespec_t {aka struct timespec}' [-Wformat=]
syslog(LOG_INFO, "Using %s device%s%s on %s in %s mode at %u baud (%u%c%u), refresh time %.3f (min. %.3f)", //
~~~^
src/gpsdclient.c:302:13:
dev->mincycle);
~~~~~~~~~~~~~
In file included from ../../lib/pud/nmealib/include/nmealib/info.h:21:0,
from src/gpsdclient.h:51,
from src/gpsdclient.c:46:
../../lib/pud/nmealib/include/nmealib/util.h:48:21: error: invalid operands to binary != (have 'timespec_t {aka struct timespec}' and 'timespec_t {aka struct timespec}')
#define isNaN(x) (x != x)
^
src/gpsdclient.c:370:8: note: in expansion of macro 'isNaN'
if (!isNaN(gpsdata->fix.time)) {
^~~~~
src/gpsdclient.c:372:33: error: incompatible type for argument 1 of 'fabs'
double fraction = modf(fabs(gpsdata->fix.time), &seconds);
^~~~~~~
In file included from ../../lib/pud/wireformat/include/OlsrdPudWireFormat/wireFormat.h:57:0,
from src/configuration.h:59,
from src/gpsdclient.c:48:
/home/hyird/lede/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-7.5.0_musl_eabi/include/math.h:203:13: note: expected 'double' but argument is of type 'timespec_t {aka struct timespec}'
double fabs(double);
^~~~
src/gpsdclient.c:390:7: error: wrong type argument to unary exclamation mark
if (!gpsdata->online) {
^
src/gpsdclient.c:457:29: error: 'struct gps_data_t' has no member named 'separation'; did you mean 'version'?
info->height = gpsdata->separation;
^~~~~~~~~~
version
src/gpsdclient.c: In function 'readFromGpsd':
src/gpsdclient.c:581:19: error: too few arguments to function 'gps_read'
gpsReadCode = gps_read(gpsdata);
^~~~~~~~
In file included from src/gpsdclient.h:49:0,
from src/gpsdclient.c:46:
/home/hyird/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/gps.h:2307:12: note: declared here
extern int gps_read(struct gps_data_t *, char *message, int message_len);
^~~~~~~~
make[5]: *** [../../Makefile.inc:352: src/gpsdclient.o] Error 1

SvenRoederer, 2020-02-21:
This seems a duplicate of https://github.com/openwrt-routing/packages/issues/547

SvenRoederer, 2020-06-07:
@hyird have you tried the patch referenced in issue #547?

closed 2020-06-13 by hyird