openwrt/routing #551: luci-app-bmx6: show the number of nodes in the network

rogerpueyo, 2020-02-21:
(From https://dev.qmp.cat/issues/339)

When more than a few nodes are in the mesh, it becomes difficult to count them. Adding a counter to the Status or the Nodes page would be helpful.

rogerpueyo, 2020-02-21:
By the way, luci-app-bmx6 is in the openwrt-routing packages repository, while luci-app-bmx7 is in the luci repository (https://github.com/openwrt/luci/tree/master/applications/luci-app-bmx7).

Which one is the best?

FreifunkUFO, 2020-10-28:
at least on luci-app-bmx7 there is "number of nodes" shown on the bmx7 status-page
[image]
p.s. bmx6 and bmx7 are not compatible to each other (to run it on one network)

BKPepe, 2023-10-07:
BMX6 was removed in https://github.com/openwrt/routing/commit/9fb9d9343ea27d6dbb5008ece10c0c843dd2c781, thus closing.

closed 2023-10-07 by BKPepe