openwrt/routing #565: BIRD1-IPv6: Static route option is unusable

xesxen, 2020-04-30:
In my OpenWRT 19.07 with luci-app-bird1-ipv6 0.3-1, the static route option (the `Routes configuration` section under `General protocols` in the UI) only allows you to enter 0-128 instead of a IPv6 subnet. This will result in a broken BIRD1 directive like 
```
protocol static {
    [...]
    route 48 unreachable;
}
```
The line to blame is here: https://github.com/openwrt-routing/packages/blob/openwrt-19.07/bird1-openwrt/bird1-ipv6-openwrt/src/model/gen_proto.lua#L225. 
BIRD-IPv4 doesn't have a datatype set, hence why it works over there: https://github.com/openwrt-routing/packages/blob/openwrt-19.07/bird1-openwrt/bird1-ipv4-openwrt/src/model/gen_proto.lua#L221

There doesn't seem to be an ip6subnet type available in luci unfortunately. Either removing the datatype line or adding a separate input for the ip6addr would solve this.

marek22k, 2023-07-21:
I think that issue can be closed due to https://github.com/openwrt/routing/issues/993. bird2 has no UCI adaoption as far as I know.

BKPepe, 2023-07-23:
Closing as bird1 was removed from the master branch because at the end of the year (2023) the support and the development will be abandoned as CZ.NIC is focusing its development on a new version (bird3).

See referenced issue.

closed 2023-07-23 by BKPepe