openwrt/routing #569: luci-app-bmx6: not working in spanshot

pedro-nonfree, 2020-05-06:
version: `OpenWrt SNAPSHOT r13124-b34165fd38 / LuCI Master git-20.126.39721-9c4f345`

- <host>/cgi-bin/luci/admin/network/BMX6/Links
- <host>/cgi-bin/luci/admin/network/BMX6/Configuration

which means that URL here is nil https://github.com/openwrt-routing/packages/blob/master/luci-app-bmx6/bmx6/usr/lib/lua/luci/model/bmx6json.lua#L49 

then https://github.com/openwrt-routing/packages/blob/master/luci-app-bmx6/bmx6/usr/lib/lua/luci/model/bmx6json.lua#L52 is incorrect and it should be (as pointed by @jow- in IRC) `if not url then`

```
/usr/lib/lua/luci/model/bmx6json.lua:53: attempt to call global 'print' (a nil value)
stack traceback:
	/usr/lib/lua/luci/model/bmx6json.lua:53: in function 'get'
	/usr/lib/lua/luci/model/cbi/bmx6/main.lua:28: in function 'func'
	/usr/lib/lua/luci/cbi.lua:66: in function 'load'
	/usr/lib/lua/luci/dispatcher.lua:1093: in function '_cbi'
	/usr/lib/lua/luci/dispatcher.lua:845: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:411: in function </usr/lib/lua/luci/dispatcher.lua:410>
```

but this works correctly

```
# lua -lluci.model.uci -e 'print(luci.model.uci.cursor():get("luci-bmx6","luci","json"))'
exec:/www/cgi-bin/bmx6-info -s
```

FreifunkUFO, 2020-11-05:
have a look at https://github.com/bmx-routing/bmx7/issues/56
and your browser-extensions-network-analysis. 
is it the "$ to %24" convertion problem?
p.s. i saw, luci-app-bmx7 is less complex than luci-app-bmx6, but almost the same code..

BKPepe, 2023-10-07:
- BMX6 was removed in https://github.com/openwrt/routing/commit/9fb9d9343ea27d6dbb5008ece10c0c843dd2c781
- LuCI-app-BMX6 was removed in https://github.com/openwrt/routing/commit/c2124dd2ad542f526ee84e1984f03dd2f5b34dfe

Thus closing.

closed 2023-10-07 by BKPepe