openwrt/routing #577: Adopt circlei from packages feed?

PolynomialDivision, 2020-06-08:
Why not using the same https://github.com/openwrt/packages/tree/master/.circleci?

SvenRoederer, 2020-06-08:
That's very similar to what @lynxis suggested in https://github.com/openwrt-routing/packages/pull/215. @yanosz PR #215 seems stalled anyway.

lynxis, 2020-07-08:
go for it!

mwarning, 2020-08-25:
@PolynomialDivision can you make a pull request?

PolynomialDivision, 2020-08-29:
> @PolynomialDivision can you make a pull request?

but someone has to acrivate that on circleci package

mwarning, 2020-08-29:
@PolynomialDivision we can setup this together on Wednesday if you like. I do not have much experience with setting this things up.  @lynxis do you want to join?

PolynomialDivision, 2020-10-05:
Or we just move this feed to packages feed as subdirectory?

SvenRoederer, 2020-10-05:
> Or we just move this feed to packages feed as subdirectory?

As discussed in #184 ?

PolynomialDivision, 2020-10-05:
Yes. ^^

PolynomialDivision, 2021-05-29:
https://github.com/openwrt-routing/packages/pull/661

PolynomialDivision, 2021-06-03:
Done.

closed 2021-06-03 by PolynomialDivision