openwrt/routing #589: [routing / bird2] RPKI does not work due to compile flag

n8v8R, 2020-08-09:
@tohojo @BKPepe 

facing the same issue as mentioned here https://bird.network.cz/pipermail/bird-users/2019-November/013959.html

caused by `--disable-libssh` in https://github.com/openwrt-routing/packages/blob/master/bird2/Makefile#L99

tohojo, 2020-08-09:
Hmm, not sure we want to pull in libssh unconditionally, though. So either it'll have to be a separate package variant, or we will need to patch bird to allow a different library or runtime detection or something...

n8v8R, 2020-08-09:
>Hmm, not sure we want to pull in libssh unconditionally, though

You consider a security risk?

tohojo, 2020-08-09:
Was mostly thinking of the size impact; don't think libssh is widely used(?), so it'll be a whole 'nother library just for that one feature (rpki) that not everyone may want...

n8v8R, 2020-08-09:
Never mind, if RPKI is not wanted on a router distro I will just fall back to compile self.

closed 2020-08-09 by n8v8R

tohojo, 2020-08-09:
Never said it would never be wanted, just said it shouldn't be unconditional. For instance, on most of the nodes I deploy Bird on, I do so to run Babel. So no need for rpki there.

If you do compile a package yourself, could you please post the size difference? Both to bird itself, and the size of the library it pulls in. If the size difference is minuscule, I won't object to just enabling it unconditionally. But if it's not (as I suspect), I'd prefer to make it a separate package variant (similar to how we have dnsmasq-full, for instance).

n8v8R, 2020-08-10:
> If you do compile a package yourself, could you please post the size difference?

Sure, if I decide to stay with *bird*, currently testing other daemons as potential replacements, e.g. _gobgp_
___
 > I'd prefer to make it a separate package variant (similar to how we have dnsmasq-full, for instance).

Probably not worth the effort since the RPKI / SLURM deployment requires certain capable hardware (mostly RAM demand > 1 GiB) that is not available for the majority of the distro's targets anyway and thus the potential user base is more of a fringe case, notwithstanding that it requires a (caching)  RPKI validator that is not available in the distro.
___
> or we will need to patch bird to allow a different library or runtime detection or something...

looking trough the source code it probably would suffice to patch

https://gitlab.nic.cz/labs/bird/-/blob/v2.0.7/proto/rpki/config.Y#L34

```
-- CF_KEYWORDS(RPKI, REMOTE, BIRD, PRIVATE, PUBLIC, KEY, TCP, SSH, TRANSPORT, USER,
-+ CF_KEYWORDS(RPKI, REMOTE, BIRD, PRIVATE, PUBLIC, KEY, TCP, TRANSPORT, USER,
```

and 

https://gitlab.nic.cz/labs/bird/-/blob/v2.0.7/proto/rpki/config.Y#L108

`-- | SSH rpki_transport_ssh_init '{' rpki_transport_ssh_opts '}' rpki_transport_ssh_check`

leaving TCP as communication channel with the RPKI validator.