openwrt/routing #609: oonf fails to build after updates September updates

SvenRoederer, 2020-10-01:
Since the changes to OONF of yesterday (30. Sept. 2020) OONF fails to build.

Most things are around

~~~
/home/runner/work/freifunk-berlin-firmware/freifunk-berlin-firmware/openwrt/build_dir/target-mipsel_24kc_musl/oonf-olsrd2-v0.15.1/src-plugins/subsystems/os_linux/os_system_linux.c:423:7: note: in expansion of macro 'OONF_WARN'
       OONF_WARN(nl->used_by->logging, "Could not drop netlink '%s' mc group: %x", nl->name, groups[i]);
       ^~~~~~~~~
/home/runner/work/freifunk-berlin-firmware/freifunk-berlin-firmware/openwrt/build_dir/target-mipsel_24kc_musl/oonf-olsrd2-v0.15.1/src-plugins/subsystems/os_linux/os_system_linux.c: In function 'os_system_linux_netlink_addreq':
/home/runner/work/freifunk-berlin-firmware/freifunk-berlin-firmware/openwrt/build_dir/target-mipsel_24kc_musl/oonf-olsrd2-v0.15.1/src-api/core/oonf_logging.h:175:7: warning: array subscript 130 is above array bounds of 'const char[70]' [-Warray-bounds]
       oonf_log(severity, source, &__FILE__[BASEPATH_LENGTH], __LINE__, hexptr, hexlen, format, ##args);                \
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~~

[OpenWrt build-log of package](https://github.com/openwrt-routing/packages/files/5314435/compile.txt)

PolynomialDivision, 2020-10-05:
That was me, sry. Something went wrong when I did this PR. You have to remove the gcc10 patch and it will build again.
I will make a PR again. Sry for the circumstances.

PolynomialDivision, 2020-10-05:
https://github.com/openwrt-routing/packages/pull/608

PolynomialDivision, 2020-10-07:
Builds again. ;) Please close.

closed 2020-10-23 by SvenRoederer